image-et-son/p600fw #8: Request for "local on/off" MIDI function as a new global synth parameter

el-folie, 2021-11-30:
"Local ON" is the normal synth state: the synth engine reacts directly to keys and wheels.
"Local OFF" is necessary when using the synth in a MIDI loopback with a sequencer to prevent double triggering the synth engine by its own keys and the same MIDI data coming back via the MIDI loop routed through a sequencer.

Some modern synths can also send out knob and switch movement via MIDI data to be recorded in a sequencer. Also in that case a "Local OFF" or "Controls MIDI Send OFF" function is necessary to prevent control conflicts with the synth engine.

image-et-son, 2021-11-30:
What happens in arp and seq mode? Should the arp and seq notes play? Proposal: yes that'll play. Arp and seq notes are not sending any MIDI anyway.

image-et-son, 2021-11-30:
The P600 only send the following MIDI signals:
- note on/off
- hold/sustain (footswitch)
- mod wheel
- pitch bend

In the setup described in the request, is that sufficient or are there any other signals needed?

el-folie, 2021-12-01:
> The P600 only send the following MIDI signals:
> 
>     * note on/off
> 
>     * hold/sustain (footswitch)
> 
>     * mod wheel
> 
>     * pitch bend
> 
> 
> In the setup described in the request, is that sufficient or are there any other signals needed?

Yes, that´d be absolutely perfect!

el-folie, 2021-12-01:
> What happens in arp and seq mode? Should the arp and seq notes play? Proposal: yes that'll play. Arp and seq notes are not sending any MIDI anyway.

Yes, as arp/seq don´t send MIDI data yet (maybe that´s an option for a later request), a running seq/arp wouldn´t interfere with incoming MIDI data. Also, in its normal "Local On" mode the P600 keyboard can be played in addition to arp/seq running, so that in "Local Off" mode this behaviour could then be retained via using a MIDI loop back, so no feature gets lost, which is great.

image-et-son, 2021-12-01:
I see no that seq and arp do send played note by MIDI. So the question is one of the two options:
- in local off mode no MIDI is sent from arp und seq; or
- in local off mode no arp or seq can play

Tough choice...

el-folie, 2021-12-02:
> I see now that seq and arp do send played note by MIDI. So the question is one of the two options:
> 
>     * in local off mode no MIDI is sent from arp und seq; or
> 
>     * in local off mode no arp or seq can play
> 
> 
> Tough choice...

Oh, yes I see that now too, which is great as it means seq/arp are MIDI-recordable in a sequencer.

Best option in this case would be to handle seq/arp the same as the internal keyboard, cutting seq/arp off from the synth engine in local off mode, so that seq/arp can still play the synth engine via MIDI loop.

closed 2021-12-07 by image-et-son

image-et-son, 2021-12-10:
Hi, I want to implement the sending of a program change MIDI message. For the local off mode I have to make a choice:

In local off mode:
1) I send a prog change MIDI but I do change the preset. The P600 will then change the program in reaction to the MIDI in program change (if connected). In this way I can MIDI record also program changes.
2) I do not send a prog change MIDI but I change the preset on the P600, e.g. I exclude preset switched from the MIDI recording in local off mode.

What would be more practical? Less confusing?

el-folie, 2021-12-10:
Good question, I can´t think of any possible scenario in my head of course, but generally I think every MIDI event shoud be sent in Local Off mode too, so that it is recordable. 

And if a Program Change event is not needed or wanted in Local Off mode, it should be able to filter it out in the sequencer program.

And if a P600 is being used as a master keyboard in Local Off mode in a MIDI loop through a PC it would be nice to still be able to send Program Changes to connected MIDI gear to change their presets.

So for now I´d think yes, being able to send Program Changes in Local Off mode would be a benefit.

image-et-son, 2021-12-10:
Thanks. But it I come across more and more questions. Currently an incoming prog change is ignored if the P600 is not in preset mode. In the moment you change a prog in local off mode you are in preset mode by design, so it's not a problem. If you are running a recorded MIDI sequence then this would only work if you activate preset mode. So next question:

1) Should incoming prog change MIDI be ignored in live also when in local off mode?
2) Should the incoming prog change MIDI activate preset mode and then change preset when in local off mode?
 
Next thing will be MIDI CC, which is currently not sent and ignored in live mode ...

el-folie, 2021-12-10:
Very good questions again! I thought about it a while...

1. Yes, I think it would be nice for the user to always have the live mode (panel mode) stable, so that no incoming MIDI program change would suddenly change what he´s doing in live mode. If he chooses to have a sequence running that also includes MIDI program changes that would mean he is in need to use preset mode anyway. So yes, ignore program changes in live mode might be a good solution.

2. That would mean that whatever the user does in live mode at the moment of an incoming MIDI program change would be overridden by switching to preset mode. I assume users might not like that but I could be wrong. 

Because of that:
Maybe again a Global Functions toggle like "Ignore MIDI program change On/Off" would be a solution for any possible case. Maybe some users would like to be sure that whatever mode the P600 is currently in that they can just fire off a sequence from a PC and be sure that the P600 reacts tp program changes. It will be hard to satisfy any user or any use case if only one specific decision is made I guess.

image-et-son, 2021-12-10:
OK, thanks, I understand now it's not really related to local off mode. I'll leave it such that prog change is always ignore in live/panel mode and I opened a separate enhancement issue for that toggle. That should be prioritized appropriately. PS: I have no idea how to implement all the additional setting yet - currently the number pad is "full"...

el-folie, 2021-12-10:
Yes, maybe it´s a good solution to simply just differentiate between live/panel and preset mode in regard to program changes. Users tend to like simple/easy to memorize solutions instead of deep menue diving. In that case a specific "ignore toggle" might not even be necessary and so space on the number pad is being saved for more important functions.