image-et-son/p600fw #15: Extend MIDI support to 7 (volume), 120 (sound off), 122 (local off) and 123 (all notes off)

image-et-son, 2021-12-02:
The refers to incoming MIDI

matrix12x, 2021-12-22:
add to midi.c around line 268 or so depending on version of code:

  else if(control==7) // added midi volume V2.24 JRS
    {
        synth_volEvent(value<<9);
    }

   else if(control==120) // All Sound off  V2.25
    {
        if (value==127)
        {
        int8_t i;
        for (i=0;6;++i)
        {
        assigner_voiceDone(i);
        }
           }
        return;

   
    else if(control==122) // All Local ON/OFF  V2.25
    {
       //0=OFF  127=local ON  - add call to code from latest version from ui.c
        
        return;
    }

	
    else if(control==123) // All Notes off  V2.25
    {
        if (value==127)
            assigner_allKeysOff();
        return;
    }

image-et-son, 2021-12-23:
Thanks, the efficient to work like this. 

About control 7: What about the function synth_refresh_MVol_Knob() you introduced? Should I be worried about this? Since there is a dedicated function for translating the MVol pot into volume (this is the 3rd cycle in synth_update()). I would now put in a check if the last active knob was ppMVol before applying the pot value. In that way, the MIDI set value can stay until someone touches the pot. But you seem to also be repairing a volume bug there, right? How can I test this bug - I have not noticed it...

Also: making master volume CC controllable sets yet another precedent: it is a knob that so far is always applied directly. Other  non stored pots are bender and mod wheel but these are are only updated internal when touched. So I would go the same way (as said above).

matrix12x, 2021-12-23:
yeah, we need to check to see if the knob or midi cc7 was the last sent volume and if the knob moved. (I did this in my version). I should have pasted the rest of the code above, but for some reason I didn't. See code below. Also, some P600s would have a drop in volume that would happen over time, unrelated to the CC7 implementation. I have not had this happen with your alphas yet, but did not do extensive testing for it. Pretty much all synths use CC7 and it may not match the physical knob position. 

synth.c needs:
near line 45 or so add:
uint16_t prevMVol,newMVol,tempmVol,

add to synth_update:
synth_refresh_MVol_Knob(); // moved for v2.33, old spot would drop in volume

add later on (I added near line 1066 or so):
void synth_refresh_MVol_Knob(void)
{
    // determine if Mvol pot was touched
    newMVol=potmux_getValue(ppMVol);
    if (prevMVol!=newMVol){
        sh_setCV(pcMVol,satAddU16S16(newMVol,synth.benderVolumeCV),SH_FLAG_IMMEDIATE);
        prevMVol=newMVol=potmux_getValue(ppMVol); //added V2.26B volume seemed to fade over time
        tempmVol=potmux_getValue(ppMVol);
    } else
        sh_setCV(pcMVol,satAddU16S16(tempmVol,synth.benderVolumeCV),SH_FLAG_IMMEDIATE);
}

void synth_volEvent(uint16_t value) // Added for MIDI Volume V2.24 JRS
{
    tempmVol=value;
    sh_setCV(pcMVol,value,SH_FLAG_IMMEDIATE);
    return;
}

matrix12x, 2021-12-30:
7 (volume) - works
120 (sound off) - works 
122 (local off) does not work...... 0 or any value really (e.g., 127) seems to set local to OFF and cannot tuen local back on with any value (s/b 127)
123 (all notes off) does not seem to be working. 

for CC 122 and 123, I'm not sure why they are not working. your code looks ok... I'll look at it again later tonight.

matrix12x, 2021-12-31:
Regarding Local ON/OFF, looking at the MIDI spec, it looks like 0 (to 63) is Local off. 127 (down to 64) is Local on.
I suggest changing the code to:

    else if(control==122) // All Local ON/OFF
     {
         if (value=<63) // local ON
         {
             ui_setLocalMode(0);
         }
         else if (value=>64) // local ON
         {
             ui_setLocalMode(1);
         }
     }

matrix12x, 2021-12-31:
regarding all notes off, the error is in my original code. A value of zero should trigger "All notes off." My error was I had a value of 127 trigger it.

Release time will still be maintained, and notes held by sustain will not turn off until sustain pedal is depressed.

    else if(control==123) // All Notes off
    {
        if (value==0) assigner_allKeysOff();
    }

matrix12x, 2021-12-31:
Regarding CC122, in UI.c this looks wrong:

void ui_setLocalMode(uint8_t on)
{
    if (on==0)
    {
        settings.midiMode=0;
    }
    settings.midiMode=1;
}

Should there not be an "else"?

matrix12x, 2022-01-02:
Alpha 10:
I confirm that CC7, CC120, CC122, and CC123 work.

closed 2022-01-02 by image-et-son