image-et-son/p600fw #16: Support control of noise level for modified P600s

image-et-son, 2021-12-02:


matrix12x, 2021-12-18:
Add to synth.c:
static void refreshNoiseLevel(void){ //for Noise
    sh_setCV(pcExtFil,currentPreset.continuousParameters[cpNoiseLevel] / 3,SH_FLAG_IMMEDIATE);
}

modify this as follows:
static void refreshPresetPots(int8_t force)
{
	continuousParameter_t cp;
	
	for(cp=0;cp<(cpCount-1);++cp)// -1 for Noise. add 1 element [cpNoiselevel] in storage.h, [enum continuousParameter_t]
		if((continuousParameterToPot[cp]!=ppNone) && (force || continuousParameterToPot[cp]==ui.lastActivePot || potmux_hasChanged(continuousParameterToPot[cp])))
		{
			p600Pot_t pp=continuousParameterToPot[cp];
			uint16_t value=potmux_getValue(pp);

			if(potmux_isPotZeroCentered(pp))
				value=addDeadband(value,&panelDeadband);
			currentPreset.continuousParameters[cp]=value;
			ui.presetModified=1;
		}
    refreshNoiseLevel();// for Noise
}

add this to void synth_init(void)
	// init

sh_setCV(pcExtFil,0,SH_FLAG_IMMEDIATE); //for NOISE stop Noise waveform

under ui.c add a parameter to whatever number you want to locate this at such as:
const struct uiParam_s uiParameters[] =
{
 /*0*/ {.type=ptCont,.number=cpNoiseLevel,.name="noise"}, //for Noise

under import.c, in the import patch section add:
    p.continuousParameters[cpNoiseLevel]=0;// for Noise

under storage.c add:
 // for Noise
            currentPreset.continuousParameters[cpNoiseLevel]=storageRead16();

under storage.h add:
    cpNoiseLevel=30, // for Noise V2.20  was CC46

note, there are some midi control panels that use this cc info to control this feature

image-et-son, 2021-12-18:
HI, do you know if there are different noise mods? Will this be very specific and only selectively applicable? And: do you know what pxExtFil is originally used for? (I could reverse engineer it, but maybe you can save me some time by putting me on the right track).

I am considering a solution in which I put the noise mod code in under a compiler flag, so it is possible to compile two binaries, one with and one without the noise, from the same consistent code base.

matrix12x, 2021-12-18:
There is only one noise mod. The mod physically rewires the filter CV jack, which originally used pxExtFil to control the filter from the CV jack. In the original P600 that external CV goes thru a VCA that is under control of pxExtFil.
The mod reuses that VCA to control to control a physical white noise source that you add to the P600.

I don't think the gligli code actually used the filter CV input. However, I don't really remember.

Here is the mod:
https://prophet600noisemod.blogspot.com/2020/08/so-i-modified-my-prophet-600-have-noise.html

The control is useable even if its not referred to as "noise."

On a stock P600, (w/o noise mod) the noise code lets the user modulate the CV input, which is really cool.

If any of that makes sense.

image-et-son, 2021-12-22:
Hi, thanks. I decided to put the CV update into the first cycle of synth_update() because it seemed to be a better suited place compared to refreshPresetPots(). I swapped cpSpread and the "new" cpExternal so that the latter is CC46 (and CC110 for fine), as you noted. I decided to keep the parameter application neutral as user without a noise mode may get confused. It's called "Ext Volt" and for the time being it will reside on 777. BTW: I cannot test it :-(

matrix12x, 2021-12-22:
I agree that is a better spot for the CV update. I put it where I put it based on a Japanese blog I read where the guy did the same mod to an older branch of the code. I didn't put too much though into whether it was the optimal spot for it or not at the time.

matrix12x, 2021-12-22:
just downloaded the current GitHub code and compiled and the Ext Volt is not working.

image-et-son, 2021-12-22:
Hm, I did not know what to expect. I measured the output of the jack with a volt meter but got nothing. Can you have a look at the commit? The essential line is 1164 in synth.c. 

Sorry, it's all a little hidden behind the changes I made for the pick-me-up mode. I have a few additional things I wanted to add plus documentation before handing it out. The pick-me-up mode is a new concept.

matrix12x, 2021-12-22:
oh, I wonder if the pick me up mode is why I'm getting some weird characters that seem to respond to knob movement

matrix12x, 2021-12-22:
sh_setCV(pcExtFil,currentPreset.continuousParameters[cpExternal] / 3,SH_FLAG_IMMEDIATE);

why the "/3"?

matrix12x, 2021-12-22:
Also, you should get zero volts on the jack. The jack is an input. If you applied a DC voltage (or an envelope voltage from a modular) to the jack on the stock unit, for example +5V, and this parameter was maxed out, you could then use this parameter to adjust the filter cutoff. 

The code changes the voltage inside the synth that controls a VCA that this external CD is fed through.

image-et-son, 2021-12-22:
Ah, thanks for the info, was not aware.

image-et-son, 2021-12-22:
> sh_setCV(pcExtFil,currentPreset.continuousParameters[cpExternal] / 3,SH_FLAG_IMMEDIATE);
> 
> why the "/3"?

That's in your code snippet :-)

image-et-son, 2021-12-22:
> oh, I wonder if the pick me up mode is why I'm getting some weird characters that seem to respond to knob movement

Yes, the idea is that if internal values and pots disagree, you pick up the value by turning the pot to that value. In order to get feedback, the display shows if you're below or above. This avoids discontinuous value changes and it allows you to dial the panel into the patch to some degree.

matrix12x, 2021-12-22:
ok, went back to my V3.34 and my noise was not working. then I opened the synth and noticed the V+ line for the noise source broke off. re-soldered. 

Your code works. I verified. sorry for bad report of bug.

el-folie, 2021-12-23:
So it looks like I need to build a noise circuit soon :)

image-et-son, 2021-12-28:
@matrix12x: Relating to the explanation of the external voltage function: do I understand correctly. that the external voltage connection to the filter cutoff frequency is hardware implemented and the voltage set in the firmware drives a CVA so that the external modulation strength can be controlled? I am wondering if we need the */3, really? Do you know why in the 2.1RC3 version the voltage was set to const. 24576?

matrix12x, 2021-12-28:
The voltage input through the external Filter CV IN gets sent to R4219 (a 47K ohm resistor) which then goes to the input of U458 (pin 9) which is a 3360. The software controls pin 10 on U458. the processed filter CV needs a non-inverting buffer (U467). This feeds another opamp which then feeds pin 15 (filter CV in) on each of the six 3372 chips.

I compiled and ran without the "/3" and it seemed to work ok. I did not detect any issues. so it can be removed.

I'm not sure about why that specific constant (2.1RC3   24576), but, it was likely just to keep the VCA "open".

matrix12x, 2021-12-28:
also, yes you understand correctly that the external voltage connection to the filter cutoff frequency is hardware implemented and the voltage set in the firmware drives a VCA so that the external modulation strength can be controlled.

image-et-son, 2021-12-29:
I have now tested it with and LFO from my modular synth. You can drive the filter nicely - that's a really neat feature. But: that parameter which in your mod sets the noise level has no effect on the external modulation strength, at least in my P600. I will probably ask in the forum but before that: have you tried this? Do you know how this should work?

closed 2021-12-30 by image-et-son