image-et-son/p600fw #17: Parameter for toggling PWM Sync Bug for backward compatibility

image-et-son, 2021-12-02:


image-et-son, 2021-12-02:
Requires #18

image-et-son, 2021-12-06:
Test Scenario: 
- turn OSC B to zero and OSC A to half
- switch on only TRI shape on both OSCs
- turn on SYNC
- switch Poly-Mod target OSC A Freq on, filter off
- turn poly-mod envelope to zero and source OSC B to half
- now observe the difference between PWM Bug on and off (Parameter 888)

el-folie, 2021-12-19:
Can´t find "PWM Sync Bug on/off" on alpha 8 to test this. Parameter 888 is "Spread" now. Has "PWM Sync Bug on/off" moved somewhere else?

image-et-son, 2021-12-19:
The parameter is on 999 now (for the time being - I image that we reshuffle the parameter in the end anyway, to make consistent groups). It's called "Sync like v2". But when I see the gibberish scrolling over the 2 character screen, it doesn't really make a lot of sense. But it's there, alright :-)

el-folie, 2021-12-19:
Thanks! And yes... ;-)

el-folie, 2021-12-19:
And confirmed the difference by the above test procedure!

As I never cared for OS versions that had the bug I never saved any sounds on those to compare. But as the parameter difference is clearly audible I´d think this issue could be seen as solved too.

image-et-son, 2021-12-23:
OK, I'll close it

closed 2021-12-23 by image-et-son