image-et-son/p600fw #22: Feature Request: Add transmitting a Program Change on MIDI out when preset is changed by keypad

matrix12x, 2021-12-03:
in MIDI.c add:
void midi_send_programchg(int8_t ProgramNumber)
{
midi_send_programchange(&midi,settings.midiSendChannel, ProgramNumber)
}

add this to midi.h:
void midi_send_programchg(int8-t ProgramNumber);

and make the call where we change the program number based on pressing the membrane in ui.c around line 736 of your current code:
midi_send_programchg(ui.presetAwaitingNumber);

matrix12x, 2021-12-04:
minor typos:
fix midi.c to:
midi_send_programchange(&midi,settings.midiSendChannel, ProgramNumber);
(was missing semicolon)

fix midi.h to:
void midi_send_programchg(int8_t ProgramNumber);
(used a - instead of _)

image-et-son, 2021-12-10:
The MIDI will also be sent in local off mode. To make that mode consistent the program change will NOT be carried out when it's changed on the number pad. Instead the MIDI in prog change message will change the program. This needs to be tested.

GrantMTG, 2021-12-10:
There must be a common standard for this. We've had "workstations" since the 90's. Maybe check out what Yamaha/Korg/Roland do?

image-et-son, 2021-12-10:
Proposal is that the program change is carried out locally. If the MIDI comes in on top of it, it will not have an effect, anyway. Also, in local on or off the rule will be same: incoming prog change will be ignored in live/panel mode and will only be executed in preset mode. There is a separate issue for introducing a toggle setting for this. If you have further insights into the "canonical" behaviour, that would be great.

closed 2021-12-12 by image-et-son