image-et-son/p600fw #23: FEATURE REQUEST

matrix12x, 2021-12-03:
Add a feature to scale detune (in unison mode) over the keyboard range, like the filter scaling. For example, as you play up the keyboard, higher notes need less detuining in unison mode than lower pitch notes.

Could prob just use the filter scaling (full, half, off).

image-et-son, 2021-12-07:
At the moment an absolute detune amount is added to each voice (multiplied by -3, -2, -1, 1, 2, 3) for all scales. Does the requirement mean that this amount should be scaled linearly with CV? Where should the pivot point be, e.g. the point which is unchanged by the setting at constant detune?

matrix12x, 2021-12-09:
Pivot point should prob be middle C. I think a linear scaling should sound fine. The idea is to have a setting that sounds good on low notes not sound like too much beating on the high notes, which is a traditional issue with unison mode on synths.

image-et-son, 2021-12-19:
From alpha 8 on detune is scaled in a fixed way (e.g. tracking amount is fixed).

el-folie, 2021-12-19:
Sounds great now as it is with fixed tracking on "full". To my ears it´s done.

matrix12x, 2021-12-23:
Sounds great as it is. I like how it is scaled by frequency.

image-et-son, 2021-12-23:
OK, we'll revisit the topic in beta testing because it changes the behaviour compared to V2.0 / V21.

closed 2021-12-23 by image-et-son