image-et-son/p600fw #24: LED blinking errors when switching to SEQ record mode and SEQ play mode

image-et-son, 2021-12-05:
1) Wanted behaviour: press RECORD (-->LED blinks) hen press SEQ1/SEQ2 (REC LED goes solid and SEQ LED blinks). Observed: REC LED continues to blink. Function is normal, though.

2) Wanted behaviour when in SEQ record mode: press SEQ button (with blinking LED) to switch from record to play mode (--> REC LED goes off and SEQ button LED goes sold). Obeserved behaviour: REC button LED does go off but SEQ button LED continues to blink. Function is normal, though.

matrix12x, 2021-12-24:
for number 2 above I think we need to set the SEQ1 and/or SEQ2 button to off by sending a set_LED so that they stop blinking  when (at around line 150 in seq.c in your current code) we are done recording notes and after we save the sequence.

led_set(plSeq1, 0, 0); //sets seq1 led to off state, no blink
led_set(plSeq2, 0, 0);

I'm sure we could put a variable where plseq1 or plseq2 is and just turn off the correct one. However, there should not be the situation where we are receding both sequences at the same time, so this may be easier coding wise. We could test to see if they are blinking and then turn them off.

For number 1 above, maybe in synth.c around line 1405 or so, we could send led_set(plRecord, 0,0); //sets record led to an off state

image-et-son, 2021-12-24:
OK, but there is a central place where all the LEDs (including the dot in the display) are set depending on the mode. This is in synth.c in the function refresh_sevenSegmets() towards the end between lines 813 and 830.  The only time set_led() is used outside this function is the tuner. I would try not to spreading out the logic further in the code. 

At the end of that block in refresh_sevenSegmets() the logic deals with the Record button. I can't understand why it is blinking after Seq1 or Seq2 is pressed because that would trigger line 824 which is never blinking. I have spent considerable time on finding out why this is not working. :-\

matrix12x, 2021-12-24:
yeah, the more I look at it, the more I think that the logic in refreshSevenSeg(void) in synth.c just needs an if then statement...

led_set(plSeq1,seq_getMode(0)!=smOff,seq_getMode(0)==smRecording);
led_set(plSeq2,seq_getMode(1)!=smOff,seq_getMode(1)==smRecording);

matrix12x, 2021-12-24:
I tried making it more explicit using if and else if (below) and had the identical results (bad) to the above two simple lines. I feel like the status bit for smrecording gets "lost." I wonder for trouble shooting if we could output via MIDI the status of the smrecording and seq_getMode(0) etc.

    if(seq_getMode(0)!=smOff && seq_getMode(0)==smRecording) // seq1 ON and record ON
    {
        led_set(plSeq1, 1, 1);
    }
    else if (seq_getMode(0)!=smOff && seq_getMode(0)!=smRecording) //seq1 ON record OFF
    {
        led_set(plSeq1, 1, 0);
    }
    else if (seq_getMode(0)==smOff && seq_getMode(0)==smRecording) // seq1 OFF record ON
    {
        led_set(plSeq1, 0, 1);
    }
    else if(seq_getMode(0)==smOff && seq_getMode(0)!=smRecording)// seq1 OFF record OFF
    {
        led_set(plSeq1, 0, 0);
    }

image-et-son, 2021-12-24:
Yes, but the smRecording is used to enter the code block lines 747-751 and that works (e.g. display shows seq count when in record mode and not when in play mode). So it cannot be fully lost. But it is wrong again for the lines 815/816 when record is switched off. Just looking at that I would assume the amRecording is not properly switched off. But then I wouldn't understand why the display doesn't just continue to show the seq count (lines 747-751). It is totally puzzling for me.

image-et-son, 2021-12-29:
Embarrassing again, like always once you figure it out. The function **led_set(p600LED_t led, int8_t on, int8_t blinking)** only ever switches blinking on but does not support switching it off. For that the LED has to be switched off and on again. I really wonder how that worked before, but I have no time for forensics :-) Problem solved, next one...

matrix12x, 2022-01-02:
Alpha 10
I confirm that the fixes worked and this operates as per "Wanted" behaviors for both 1 and 2

closed 2022-01-02 by image-et-son