image-et-son/p600fw #26: Pick-me-up control mode in preset mode

image-et-son, 2021-12-06:
At the moment in preset when you touch a switch or pot the value read form it will automatically be applied. However, it would be much nicer and practical to have a control mode variant in which you need to turn the pot or switch to the position corresponding to the preset and "pick it up" from there. The display could indicate whether the preset value is UP or DOWN from the current pot position (once the pot is touched).

image-et-son, 2021-12-11:
Changes need to be implemented in refreshPresetPots() in synth.c (in addition to flag indicating the status (less, greater and equal = picked up). What would it be for switches? just: unequal and equal = picked up. Except for the three way switches either it is already picked up or will be as soon as the switch is moved, so difficult to solve.

el-folie, 2022-01-05:
Tested for knobs - works fine! Also like the slow sweeping necessary to pick up the current value.

The switch question is tricky. Though to me switches are stationary functions, so I think it would be logical to just instantly pick up a switch setting like in panel mode.

image-et-son, 2022-01-05:
Maybe: if switched away from the current value do nothing?

image-et-son, 2022-01-05:
The pick-me-up for switches will be continued in #62

closed 2022-01-05 by image-et-son