image-et-son/p600fw #27: Reorder the add. patch parameter layout

image-et-son, 2021-12-07:
At the moment the aad. patch parameters are distributed in a non-intuitive way on the number pad. With this enhancement the parameters belonging to a specific group should all be on single number, while pressing the number cycles through the parameters.

First idea for grouping:

0 | Speed
1 | LFO
2 | Vibrato
3 | Mod Wheel
4 | Noise
5 | Envelope
6 | Bender
7 | Note Play
8 | Oscillator
9 | Velocity

image-et-son, 2022-01-13:
This would be potential layout, also adressing the issue of switching between different layouts:

[image]

el-folie, 2022-01-13:
Nice logical layout! And I so very appreciate the idea of original/GliGLi panel switching. 👍 

Just one thing, and I guess Matrix12x would also like this: As Ext Voltage controls the volume of NOISE (when added as a mod), it becomes potentially an oscillator level. So Ext Voltage might better be on a first press than on a second one for easy sound editing.

matrix12x, 2022-01-13:
I think its a ver good layout. 

I am of the opinion that LFO/vibrato target should be on the same level press. But I see why you didn't put it there. It makes sense to access speed then amount/depth. 

Also I would like to advocate for the following renaming/relabeling: 
"Amount" should be "depth" .... although the front panel calls the LFO modulation depth "amount" so I get it.
"Routing" should be "Env Routing".... however, it is in the Env column.. so maybe its obvious

At some point we should add the various shapes to the vibrato and have the Menu options accordingly. I'm not sure if this should be planned for.

el-folie, 2022-01-14:
Just a little detail:

Proposal for change of order/labeling of button 5 Envelope:
1. FLT ENV SHAPE (because it´s the 1st/higher on the panel)
2. AMP ENV SHAPE
3. Routing

closed 2022-01-20 by image-et-son