image-et-son/p600fw #30: Proposal for a smooth logarithmic curve mode for the mod wheel

el-folie, 2021-12-08:
As said on gearslutz by user Mastropiero the P600 mod wheel response is quite like "on or off" because the onset of modulation is too intense with the standard linear function.

Proposals were either a toggle for lin/log to also keep an option for original behaviour or just a smooth log curve for the mod wheel response.

Decision is up to the coders ^^ ;-)

el-folie, 2021-12-12:
Enhancement confirmed in alpha 6: Smooth onset of modulation by mod wheel now possible. :)

image-et-son, 2021-12-13:
Next alpha will contain two variants of this so that a direct comparison is possible. A final decision will then be made. So -> issue back to "in Work"

el-folie, 2021-12-19:
Sounds fine now! Enough variations to let the user choose. 

Maybe the old "Hard" ones could even be erased, Smooth is always better.

closed 2021-12-19 by image-et-son