image-et-son/p600fw #34: Misc. setting button "7" does not work with "From Tape" held done, only in shift lock mode

image-et-son, 2021-12-11:
Pressing the button "7" while holding "From Tape" down does nothing. But all other numbers work and access the respective misc. setting correctly.

matrix12x, 2021-12-11:
I had this issue also. Never was able to solve it. and I was positive I coded it correctly

image-et-son, 2021-12-12:
Maybe it's a hardware thing. There must be a reason why GliGli crammed in Spread and VCF Limit into number 9 and leaving 7 unused. I will make some more tests to confirm. BTW: I am thinking about making Spread a patch parameter.

matrix12x, 2021-12-12:
I agree, spread would work better as a patch parameter instead of global

image-et-son, 2021-12-12:
I logged the button status to analyze the situation. It's interesting. When I hold "From Tape" and then press, say, "1", then release "1" and then release "From Tape" I get the following sequence of events:
1) "From Tape" pressed
2) "1" pressed
3) "1" released
4) "From Tape" released.

If I do the same thing with "7" I get:
1) "From Tape" pressed
2) "From Tape" released
3) "From Tape" pressed
4) "From Tape" released.

This also explains why I sometimes go into shift lock mode when I try to use the "7". I would now conclude that it is a hardware design error. The number "7" should not be used for the settings menu, I think, because that different behaviour is not user friendly. I'll find a place to document it for everyone.

closed 2021-12-12 by image-et-son

matrix12x, 2021-12-12:
I'm not 100% sure its a hardware issue. On the schematic "from tape" and 7 share a logic line to D108. However, "to tape" and 6 share a line and they do not seem to have this issue. 
"from tape" = from pin 9 on U104 to D108 to pin 10 on U106 which feeds line DB7
"7" = from pin 10 on U104 to D108 to pin 10 on U106 which feeds line DB7

Maybe its the sensing code for "from tape" once "from tape is pressed?

image-et-son, 2021-12-12:
This is a very good point! I immediately checked "To tape" and "6" as well, and they behave exactly the same way. In short: it seems that for these pairs of buttons you cannot have them both in the status "pressed" at the same time and pressing one will instead only release the other. Only for "To Tape" and "6" we don't use that combination in the UI. There is nothing we ca do about it in the firmware. 

I'd take back the term "error", though. The design was not made for this,  Are there any other limitations like that which you can see?

matrix12x, 2021-12-12:
None that I'm aware of. 
It's funny, I spent hours trouble shooting that in code a few years back and never had a solution and never thought to look at the hardware until you said it was a hardware bug. Now that we went over this above, it seems super obvious that the hardware design cannot absolutely was not designed for this. lol