image-et-son/p600fw #38: "Hold Pedal" not working properly with Unison

image-et-son, 2021-12-13:
From el-folie on Gearspace (https://gearspace.com/board/showpost.php?p=15763038&postcount=2999):

1. In panel/live mode, Local On:
- No unison: it works as it should
- Unison:
a. If you switch to Unison and press a key and then activate the hold pedal then the next key pressed will not sound in unison anymore. Also there is no hold function.
b. If you switch to Unison and first activate the hold pedal and then press a key it will sound in unison but there is no hold function.

In all cases the hold pedal sends MIDI data on/off though.
It´s the same behaviour in preset mode.

2. In panel/live mode, Local Off:
- No unsion: it works as it should
- Unison:
a. If you switch to Unison and press a key and then activate the hold pedal then all the next keys pressed will not sound in unison anymore and the first pressed key will keep sounding as long as the hold pedal is active/on (together with the last key held). Very funny behaviour.
b. If you switch to Unison and first activate the hold pedal and then press a key the behaviour is correct. Keys are held and sound in unison.

image-et-son, 2021-12-13:
Now the sustain is not carried over into unison mode. Also, in unison mode pedal MIDI events are not sent. The arp overrides the pedal function. It becomes a start stop pedal.

image-et-son, 2021-12-14:
Another bug: when you hold down a chord in unison mode, then press the pedal to latch it an then release on key the chord will be triggered if and only if the key released changes the priority to another key (e.g. when priority is "low" then releasing the lowest key in the chord has that effect). Releasing any other key does nothing. While this behaviour is strange and unwanted anyway, there is also the bug that the release not never receives a key off event, so will sound indefinitely  or until the assigned requires that voice or unison is switched off.

el-folie, 2021-12-14:
Just some thoughts...

If the "change-chords-on-the-fly" behaviour by pedal is so clumsy and unstable (and I think I´d never use that feature as it was already clumsy on the Sequential OS) and Mastropiero as a live keyboarder said he´d use a conventional hold pedal function in every thinkable use case - then maybe the implementation of just a simple conventional hold pedal feature like already in poly mode would be best for normal users?

I can understand if you´d feel like fixing all the bugs to make it work in the way you´d like to. I´m just thinking of saving some energy and "brain power" if it´s extremely complicated and maybe not worth the outcome. At the same I think most users would just expect to hear a unison sound when switching to unison and then use the pedal conventionally like they would in poly mode.

matrix12x, 2021-12-23:
Hold pedal seems to work as expected. 
No MIDI on local off
MIDI cc64 on Unison off, local on
Unison on, the pedal acts as a chord latch

Is there anything else about it I should check?

closed 2021-12-23 by image-et-son