image-et-son/p600fw #39: Add MIDI CC implementation of new parameters after v2.1

image-et-son, 2021-12-13:
- Envelope Routing
- Sync Vintage
- Spread

matrix12x, 2021-12-20:
By adding the parameters to storage.h they are implemented as CCs. Heads up because they are spaced so hold pedal is 64

matrix12x, 2022-01-01:
Verified that MIDI CC for spread works.

In Preset mode, when "to tape" is showing the current value of a parameter and we change that parameter via MIDI, should we update that display to show the current value?

image-et-son, 2022-01-01:
The display shows the last touched pot, so the user knows what the value shown refers to. When sending MIDI it is not clear what value is shown. Also if you change several parameters at the same you'd get a lot of gibberish on the display. The only thing I could imagine is updating the display value that is already showing (last touched pot) if that is changed by incoming MIDI CC:

matrix12x, 2022-01-01:
I think updating the display value that is already showing (last touched pot) if that is changed by incoming MIDI CC is a good idea.

image-et-son, 2022-01-01:
I'll track that idea in a separate issue. I am fixing the MIDI CC bugs for alpha 10 to that this ticket can be closed. 

This display idea will need to come later.

matrix12x, 2022-01-08:
Where is the best spot to find a list of the new parameters that have MIDI CC implemented?
I want to go through them for testing.
I believe it is:
	spEnvRouting= CC74			
	spLFOSync= CC75
	spPWMBug= CC76

and spread which I already verified.

image-et-son, 2022-01-08:
Hi, here is an extract from the user manual I have been working on:

[p600fw_User_Manual_ReferencePages.pdf](https://github.com/image-et-son/p600fw/files/7833925/p600fw_User_Manual_ReferencePages.pdf)

image-et-son, 2022-01-08:
Without shuffling things around there is no room anymore for continuous parameters. I found that the MIDI CC implementation of he P600 already deviates from standards, so if it comes to it, maybe it would be time to move some parameters to other places.

matrix12x, 2022-01-08:
I confirmed the following CCs work:
EnvRouting= CC74
PWMBug= CC76

I'm not sure how LFO sync works, so I didn't know what I was testing for.

image-et-son, 2022-01-08:
Hi, thanks for testing, as far as LFO Sync is concerned, this is the paragraph from the upcoming user manual:

[image]

matrix12x, 2022-01-09:
Thanks for the explanation.
I confirmed MIDI CC works for LFO Sync. CC 75.

I think that's everything. If I missed testing a CC parameter, let me know.

closed 2022-01-09 by image-et-son