image-et-son/p600fw #41: Proposal for seperate "unison detune" and "poly mode detune" values storable per patch

el-folie, 2021-12-15:
The "Spread on/off" toggle wouldn´t be needed anmore.

- Unison detune usually needs higher values for a nice thick sound
- Poly mode detune needs very low values for a subtle "slop" but still musical sound
- Therefore two storable detuning values per patch (1. unison - 2. poly mode) would be nice as to be able to switch from poly to unison and back with nice preprogrammed detunings each.

If feasible even a seperate third feature could be derived from the then dropped "Spread" parameter by making the envelope variations, which until now were very subtle or maybe only masked by the unison detuning, an own seperate patch parameter with greater variances (so, 3. env variation). The question is how GliGli programmed the envelope variation, either as being static (on/off by Spread on/off) or dependent on the unison detune value. Maybe this can be seen in the code?

These are just proposals/ideas still being discussed on gearslutz.

image-et-son, 2021-12-15:
The current implementation of spread with respect to envelopes is static: for each voice i the add-on to all eight envelope parameters is: spread=((1+(i>>1))*(i&1?-1:1))<<8; In plan English: there is an add-on of the six voice with factors (-1,+1,-2,+2,-3,+3) but this is still scaled down by a factor of 256 compared to the full range of values of those parameters.

el-folie, 2021-12-16:
Thank you for investigating the code and translating it for mere mortals.

So it means that the current Spread implementation distributes the (static and tiny) envelope variation symmetrically between the detuned voices (which makes perfect sense for unison).

There´d be room for improvement:
- Seperation of envelope variation from detune
- A continuously variable depth of the still static variation (up to perceivable chaos) 
- A continuously variable depth of a more complex variation behaviour, which would involve first investigating how analogue envelopes (OB-X, JP-4, etc.) actually behave. If interested in that I´d offer to read what I can find about it and report back.

image-et-son, 2021-12-16:
I would propose the following: replace the spread switch by one continuous spread parameter which introduces detune per voice, detune of cut-off and variations in ADRS parameters. I'd bundle that together into one parameter in a meaningful and balanced way (with a few test iterations to get it right) rather than have it separately customizable. I really worry that we're making this too complex for normal users, whom I want to reach with the release.

Further, I propose to keep the existing detune as it is (only applied in unison) but to apply the above spread on both, the unison and poly mode.

el-folie, 2021-12-16:
You´re absolutely right by all accounts!

It´s just better than any attempt to dissect GLiGLi´s approach into several parameters. Your proposal would be comparable to the Prophet 5rev "vintage knob" - much more simple and intuitive for the users.

Thank you for even looking into this "parameter mess" ;-)

el-folie, 2021-12-17:
Maybe one "variance" knob still could simply have different zones like this:

1. 0 = off
2. 1-24 = offset detune only - from min to max
3. 25-50 = env variation only - from min to max (levels a lot less than times)
4. 50-74 = offset detune & env varation - from min to max
5. 75-99 = offset detune & env variation & cutoff varation - from min to max

That way, the knob would let the user have the benefit of distinct decisions. It´d pack a lot of functionality into one knob while still being simple to explore by just twisting one knob.

el-folie, 2021-12-30:
Found a bug in alpha 9: Returing Spread to 0 after 99 doesn´t bring the attack times back to be perfectly aligned. So it seems that Spread keeps a difference at 0 if once being turned up.

image-et-son, 2021-12-30:
OK, tracked in issue #53

image-et-son, 2022-01-08:
Hi el-folie, apart from the separate bug #53 , do you have an opinion regarding the spread function as a whole? The strength, the relative effect on the different components? In the first comments you also mentioned that you notice some "patterns". It is true, that currently for each voice the effect on tune, filter tune A+B, attack, decay and release all have the same direction....

el-folie, 2022-01-08:
Honestly, I haven´t had the time yet to do thorough musical testing of the Spread variations so far. One thing I already can say though, is, that the variation of cutoff differences by envelope seemed to samey, as in high-low-high-low-high low, or like +1-1+1-1+1-1. While that pattern still could be possible result of deviation after years of use it sounds too regular to my ears. More natural would be something like: 0-(+0.2)-0-(-0.1)-(-0.2)-(+0.1). So, with the values just more evenly spaced out while still being closer together. With Spread all up values being wider apart of course.

But to really say something useful. I´d need to do in-depth testing. Right now I´m fighting with Pro Tools to record Repro-5 envelope shapes... ;-)

el-folie, 2022-01-14:
Now I know what´s throwing my ears off with Spread (and yes, like in the new table, "Vintage" would be a cooler label):
As Vintage is supposed to mimic vintage behaviour, it doesn´t feel right that at higher values ONLY the octave scaling of individual OSC pairs (= voice) gets wonky. At that kind of wonky on a real vintage synth usually the detuning between OSC pairs on a voice is more off from voice to voice than the fundamental octave scaling, which usually stays intact.

Proposal:
A maybe simple experiment would be to let Vintage affect only one ocillator in each voice´s pair instead of both, so that the "off scaling" introduces the detuning in a voice AND the octave scaling. Maybe that´ll sound more natural for the whole Vintage knob travel.

el-folie, 2022-01-19:
Tested - works! 

Sound-wise a whole lot better than before, also the whole range of Vintage intage is now usable. I´d say done for now.

closed 2022-01-19 by image-et-son

matrix12x, 2022-01-20:
Tested Alpha 12 Vintage
Sounds awesome!!!