image-et-son/p600fw #42: Proposal for a continuous linear detuning patch parameter. FINE tune then becomes beating speed control

el-folie, 2021-12-15:
Normally on a V/oct synthesizer, the beating frequency between detuned oscillators A and B doubles by octave, quadrupels by two octaves, octuples by three octaves, and so on. This behaviour leads to sounds that may have a nice slow beating on low octaves but fast detuned beating on high octaves on a 5 oct keyboard. A linear detuning parameter can help to equal the beating frequency in low and high octaves, so that the patch sounds nice over the entire keyboard range.

This could be realized by a scaling parameter for the FINE tune knob on OSC B:
- knob at 0 = normal exponential behaviour on beating by detune (equals stock P600 mode)
- knob at 99 = exact linear detuning, so that the FINE tune knob effectively becomes a beating speed knob

By introducing this parameter also the poly mod section would gain from it:
- as the beating speed now can be deliberately set, very slow poly modulations to OSC A are possible over the entire keyboard range.
- also, the distribution of the modulation speed over the keyboard range can be deliberately set by the linear detuning parameter (and not only follow a fixed exponential scaling).

This way the patch programmer has total control over beating frequency and poly modulation speeds over the entire keyboard range, even scalable by ear by a continuous parameter for the best sounding outcome.

As a side note, by this feature we´d in a way get back the lost "Lo Mode" feature from the Prophet 5 which had been dropped for the P600. We´d get the slow poly modulations by OSC B. We of course still won´t get a "Lo Mode OSC B" to audibly mix back into the sound.

image-et-son, 2021-12-15:
OK, the one thing I don't understand yet is what to do if the frequency difference between the two oscillators is large, even in poly mod. E.g you could have fifths + 1 octave between them. I do this a lot. The cross-modulation creates complex harmonics. I know that you can adjust bad sounding keyboard ranges by slight fine tuning as you play higher or lower notes, but would a linear shift on the fine tuning be able to fix that across the entire  keyboard (or a large part of it) also in this case?

el-folie, 2021-12-16:
A simple answer to your question is this (if I understood your question correctly): 

Any >speed dependent movement< in detuning or harmonic content will be influenced by the linear detuning parameter in that the speed of the movement will be more controllable, like for example made slower over several octaves. But the change in speed of the harmonic movement will NOT change the >frequency of the harmonic content< or the >composition of the harmonic content itself<. 

So in a way we´d get comparably slow movements like in Prophet 5 "Lo Mode" but we´d still not get a real pitched down OSC B to excite a different harmonic content by poly mod or to have LFO like response. In order to do that we´d need OSC B to pitch down 4 octaves to reach LFO speeds. I don´t know if that´s possible by programming but would assume that the lowest C on the P600 is not set at 1 volt. But if it were there´d at least be three octaves of possible pitch down by code (if my assumption 0.25v/oct is correctly remembered from the service manual).

image-et-son, 2021-12-17:
Result of analysis: the voltage shift function for detune d as a function of voltage v should be  delta-v= 0,5V * log2( 1 + d/f(v)) where f(v) is the frequency (as a function of voltage). Then the shift in frequency is linear.
[Voltage_to_Frequency.xlsx](https://github.com/image-et-son/p600fw/files/7737792/Voltage_to_Frequency.xlsx)

el-folie, 2021-12-18:
I looked through the table, a lot of work! It´s great to see that you´ve cracked this riddle!

So if I understand correctly, a continuous parameter to smooth out the FINE knob travel towards a linear detune function would mean that the usual voltage addition by the FINE knob gets interpolated gradually from value 1 to 99 with the function you developed above. At value 99 of the continuous parameter ("detune tracking") FINE would then become a "beating speed" control. And all the values from 1 to 98 would allow for a fine control of poly modulation movement speeds and beating speeds of simple oscillator-detuned sounds.

Looking totally forward to this. It´d be a unique feature in vintage analogue poly land for sure...

image-et-son, 2021-12-18:
Yes, but there are questions: In the present function the fine frequency dial for OSC B tunes 1 semitone up or down across the entire frequency range, so the "unit" of that dial is "semitone". If you make it a linear shift then the "unit" of the dial is frequency. In this case I need to make a decision about the scale, e.g. on which note will the full right position of the freq B dial be the same for both variants? C3? C4? A4?

I am also looking forward to this. I consider this a real experiment as I have never seen this anywhere.

el-folie, 2021-12-18:
IIRC the late Chroma Polaris had a linear detuning parameter, so maybe the big Chroma had it too. But apart from that nothing in vintage synth land had it. So, it´d be breaking into new territory with a P600.

I need to put some brain-time to the semitone-linear scaling question...

image-et-son, 2021-12-18:
This does not contribute to the discussion at hand but I need to say that the Chroma Polaris is very high up on my list for next acquisitions. Your information only confirmed this.

el-folie, 2021-12-18:
Well, the Polaris sure sounds nice. I had it on my list too but when I got the P600 my vintage poly lust finally became satisfied. The Polaris can receive poly aftertouch, another unique function, so def worth an acquisition.

el-folie, 2021-12-18:
About your question:

In my simple understanding the FINE knob is bipolar and could stay bipolar. So it could just behave the same way towards + and - when only its function becomes more linear. When the additional parameter "tracking detune" reaches value 99 the FINE knob would react so, that at zero (middle position) the beating stops (like with detuning). And when moving to + the movements starts to go in one direction. When moving to - it moves in the other direction. But the speeds of movement would be the same at full + or full -. By working this way FINE at zero would be zero for every note on the keyboard, likeweise if FINE would become a bipolar linear detuning control (at value 99 of the detune tracking parameter).

But maybe I didn´t fully get the question?

matrix12x, 2021-12-23:
I think we could probably do something similar to what is now done with detune, where it is scaled based on freq.        detune=(1+(v>>1))*(v&1?-1:1)*(detuneRaw>>8);           // scale detune with tone, e.g. less pronounced at higher frequencies

I think the fine knob value is currently added to the CV value for the oscillator. so we could have a mode where the knob is scaled by freq before being added.

by tweaking this line in synth.c based on which mode (beating speed control/fine):
fineBFreq=(fineBFreqRaw>>7)+INT8_MIN*2;

or this ,line (prob better off on the line above):
        cvb=satAddU16S32(tuner_computeCVFromNote(BNote,baseBPitch,pcOsc1B+v),(int32_t)synth.benderCVs[pcOsc1B+v]+mTune+fineBFreq);

el-folie, 2021-12-23:
Would it then be like a switch between FINE or Beating Speed, or could the "scaling to frequency" be realized in a way that an additional parameter would allow for the Beating Speed to be continuously variable?

image-et-son, 2021-12-23:
First, I think that in contrast to the detune scaling it will be necessary, to calculate the exact voltage for the desired shift because my gut feeling tells me that a linear approximation is not enough. But that will not be a problem. The next thing could be that it is also necessary to take tuning into account, but my proposal would be to try it without first and then decide if the effort is worth it. 

As far as the continuous parameter is concerned, I think that "fine tuning" (as it is now) and "beat intensity" (to be programmed) are two entirely different functions, one being in semitones, the other being in absolute frequency. I'd propose to have a switch. Or do you want to explore the world in between? I often simply want to get rid of the beating variation across the keyboard. What do you think?

matrix12x, 2021-12-23:
I think you are correct, I think they should be two different functions.
My personal goal, as well, would be to eliminate beating variations across the keyboard.

el-folie, 2021-12-23:
Okay, I understand.

Just to explain what went on in my head:
My original idea was a function to be able to make simple detuned notes in higher octaves slower in beating by having a continuously variable parameter in addition to the FINE knob. That way every beating speed between normal expo detuning by FINE and coming to a (nearly) full stop of beating at linear detuning would be possible to be able to find the best fitting musical amount of beating for a given sound across the keyboard. Additionally I thought that that function would also bring great benefit for poly mod to be able to have complete control over the variation of motion speed by poly mod across the keyboard, from nearly a full stop for the whole keyboard to a variation from slow at C0 to somewhat faster at C5. Due to the analogue nature of the beast the varation will still be quite lively and drift over time (I assume). 

I agree that a linear switch would bring great benefit for sound creation. So testing the switch first and seeing then if a continuous parameter would bring further benefit surely is the best approach.

el-folie, 2022-02-01:
I can honestly say this now:

As the fatastic new fine-tune delivers dead slow speeds on a warmed up machine this feature has diminished my wish for a linear-detune parameter. I´m aware that linear detuning still is something different, but the new slow detune speeds are good enough for me to get a somewhat similar effect. 

So, in essence, to lessen the "feature backlog", we could probably drop linear detune from the list, unless you still see more potential in it and therefore would like to pursue it. I just wanted to say that I´d be completely okay with dropping it.

image-et-son, 2022-02-01:
Thanks for the feedback. I found it difficult to implement but it is conceptually not impossible. My goal would be to get this version out in the world on a broad scale - I do think that there is something to be learned in the process, that means to think about what makes a reliable but also inspiring instrument for a broader audience. I believe that this create significant new interest in the P600.

After that I can see many potential future additions, including this function. For me "backlog" means that this will be assessed / prioritized later and backlog items may or may not be selected. I have not deleted any of the proposed changes, even though I am sceptical of some of them. So it is not harm to leave it in there and talk about it later.

el-folie, 2022-02-01:
You´re welcome! And thank you too for explaining your view on features and vision for the P600. And I agree completely. The new features will make the P600 a much more desirable instrument for the future, it already is, thanks to GliGli, you and all the coders who contributed to this journey.

matrix12x, 2022-02-02:
@image-et-son I agree, I think for the remainder of the Alpha, it should just be bug squashing. 
New features should be for the next actual revision.