image-et-son/p600fw #43: Feature Request: setting number of unison voices from between 2 to 6 voices.

matrix12x, 2021-12-16:
I think a cool feature would be to be able to set the unison voices from 2 to 6 voices. This should stored with the patch. The in effect you could have a poly unison. 

At 2 voice unison it would be 3 note polyphony, and at 3 voice unison it would be 2 note polyphony. At 4-6 voices, it should be mono unison.

image-et-son, 2021-12-19:
Thought about it again. I only really know this feature in poly mode, e.g. sacrifice half of the voices for a thicker sound. In the case of the P600 I am not sure how to reconcile this with legato in unison mode. From the Jupiter 4 I know a mixed approach, unison 2: it plays as many voices per note as possible, e.g. one note played = 4 voices, when playing a second note and it will play both with 2 voices and so on. But there is not legato. 

So I think, the "best" one could consider is a doubling switch for the poly mode.

matrix12x, 2021-12-20:
What about setting unison to between 2, 4, or 6 voices, in unison mode only. Still mono (not poly).

Although, I do like your doubling switch idea.