image-et-son/p600fw #44: Support MIDI patch load into panel controls (in preset mode)

image-et-son, 2021-12-16:
Currently any MIDI patch SysEx ist written to the storage (of the number in the SysEx data). The proposed change is that Prophet 600 accepts MIDI storage only in a dedicated patch receive mode and that without this mode, e.g. in preset mode incoming MIDI patches are loaded into the active panel controls only. In live mode MIDI patch data should not be not accepted.

matrix12x, 2021-12-16:
What about receiving patch sysex in manual mode stores the patches in the location indicated in the sysex. in preset mode the singe sysex patch is loaded into active panel controls only. Now, what to do about a bulk dump? I guess the last one sent is in the active panel.

image-et-son, 2021-12-16:
I thought about it. But for the same reason that GliGli excluded MIDI CC control in live/manual mode, I would not accept MIDI SysEx in live/manual mode: it causes conflict of pots/switch positions and control values. At the moment there is clarity that when the preset LED is out, you get what you see and I think that is good. My solution: we create a new status (for example by "From Tape" +" Record") in which the P600 accepts SysEx into storage (bulk, multiple, single). I think this is sensible anyway, as it protects your stored patches against unwanted/accidental MIDI SysEx.

matrix12x, 2021-12-16:
I agree that your proposed solution above is a good idea, but I think 100% that unsophisticated users will have problems and wonder why their bulk dumps don't work.

closed 2021-12-22 by image-et-son