image-et-son/p600fw #45: Feature Request: MIDI reception of aftertouch (channel pressure) and poly aftertouch

matrix12x, 2021-12-18:
I added this to my branch of the code, but never implemented the note selectivity of poly aftertouch. I treated it like a mod wheel event. IN theory it could have it's own targets separate from mod wheel targets.

add to midi.c :

// aftertouch JS August 2020 note: need to add note selectivity
static void midi_aftertouchEvent(MidiDevice * device, uint8_t channel, uint8_t note, uint8_t value)
{
    if(!midiFilterChannel(channel))
        return;
#ifdef DEBUG_
    print("midi poly aftertouch ");
    print(" value ");
    phex(value);
    print("\n");
#endif
//    synth_wheelEvent(0,value<<9,2,0);
    synth_wheelEvent(0,value<<9,2,0,0);
}

// aftertouch JS August 2020
static void midi_chanpressureEvent(MidiDevice * device, uint8_t channel, uint8_t value)
{
    if(!midiFilterChannel(channel))
        return;
#ifdef DEBUG_
    print("midi chan pressure ");
    print(" value ");
    phex(value);
    print("\n");
#endif
//    synth_wheelEvent(0,value<<9,2,0);
    synth_wheelEvent(0,value<<9,2,0,0);
}

el-folie, 2021-12-18:
(And polyAT for keys routed to voice filter cutoff and vol level plus polyAT translated to monoAT for the mod wheel would bring us near CS-80 behaviour.)

image-et-son, 2021-12-19:
That opens up the need for a lot of parameters, e.g. the above would already be two more continuous parameters. Before we head into this I would like to get a feeling what the likely final list of targets is? 

BTW: what is the reason to translate to monoAT for mod wheel? It would be easy to apply polyAT to the mod wheel effect.

el-folie, 2021-12-19:
Ah okay... and totally agreed, many features are in discussion already.

Maybe I´m wrong about that assumption, justsome thoughts: 
On a CS-80 with 8 voices there´s only 1 LFO, not 8. So the polyAT of its keyboard must somehow interpolate the different key polyAT values to one smooth monoAT signal to control the one LFO. I´d think that if 8 discrete polyAT values would simultaneously control the single LFO - that the result would maybe sound quite choppy and not smooth like it is on a CS-80. 

But maybe there´s just some kind of lag filter to slow down the control voltage generated from the 8 polyAT voltage streams just enough to smooth out the LFO modulation. Whatever they did it feels like smooth monoAT to LFO when playing a CS-80 VST via an ELKA MK55 polyAT MIDI keyboard.

el-folie, 2021-12-19:
I forgot to address the modulation targets.

If CS-80 behaviour would be the goal, polyAT targets are: 

- voice brilliance (= voice cutoff)
- voice level (for the P600 this would be the cem3372 final VCA gain CV)
- mod wheel

On a CS-80 both speed & depth of the LFO can be controlled by AT.
So, if the P600 mod wheel could also control speed & depth at the same time, we´d be there.

image-et-son, 2021-12-19:
Ah, if you control the speed of the mod wheel (which is either vibrato or LFO in our case), then you cannot do it polyAT, as we only have on synced LFO for all voices. (BTW: I have a ´n Ensoniq ESQ1 which has a separate LFO for each voice. For pitch that's chaos but the ultimate modulation target for that is: panorama position, which the ESQ1 supports!) 

How can the CS80 control speed and depth? Does it have two controls for that?

el-folie, 2021-12-19:
"How can the CS80 control speed and depth? Does it have two controls for that?"

Yes exactly!

The CS-80 has Aftertouch-controlled levers for Speed, VCO and VCF modulation depths. Speed by AT sets the depth of acceleration, VCO and VCF the usual modulation depth of frequency and brilliance, while the speed control in the "SUB OSC" (=LFO) section sets the general speed. Ancient tech... ;-) I´d love to have such a thing, but it´s unobtainium due to price nowadays. Vangelis and Blade Runner in 1982 totally mesmerized me in the cinema...