image-et-son/p600fw #48: Support single patch MIDI dump from the P600

image-et-son, 2021-12-23:
Currently, the only way to dump a single patch via SysEx from the P600 is sending a dump request SysEx to the P600. Fomr the panel/UI only patch bank dump is supported (currently FromTape+6). One enhancement would be to make it possible to select a single patch number to be dumped.

matrix12x, 2021-12-23:
Maybe we could have "from tape" + "Seq 1" transmit the current preset via MIDI. I feel like "from tape" + preset would make more sense, but I think you recently used that for default patch.

in UI.c under switch(button) (near line 64) do this (note: need to put the current preset number in the variable):

case pbSeq1: // dump current preset
                midi_dumpPreset(need to put the current preset number here);
		sevenSeg_scrollText("current preset dumped",1);
		refreshPresetMode();
		refreshFullState();
                return 1;

matrix12x, 2021-12-30:
I confirm single patch dump in Alpha 9 works. I like that the solution was entering the patch number.

closed 2021-12-30 by image-et-son