image-et-son/p600fw #49: Make seq and arp controllable by MIDI / external keyboard

image-et-son, 2021-12-28:
Currently the sequencer and arpeggiator only accept input from the internal keyboard. The enhancement is to make it possible to accept MIDI notes as well. 

This would be very important for users who have hardware problems with the built in keyboard.

image-et-son, 2021-12-30:
From Jpro600k on Gearspace:

I have confirmed that both the sequencer and the arpeggiator receive MIDI signals in the DAW.

1. Unable to start sequencer from DAW when set to MIDI SYNC.
2. The sequencer cannot wait to start. Is it possible to receive the start signal and start playback?
3. Does Prophet-600 not receive the playback speed from the DAW?

matrix12x, 2022-01-02:
How should/would the P600 interpret the MIDI start realtime message?

I.e., what would start? arp? (which one) or Seq (which one, 1 or 2?)

image-et-son, 2022-01-02:
I don't know. Maybe: in sync ext the seq/arp never start from the buttons but go into wait? So the user chooses by activating the buttons. Should toggling of arp hold be supported? Toggle of seq record mode?

matrix12x, 2022-01-02:
I think it should work as follows:
In Sync Ext, receiving MIDI Start real time message should start the last Seq that was pressed. If no Seq button was pressed, then it should start Seq 1.

I don't like the idea of going into a wait.

Maybe song select (System Common Messages) should be supported to select Seq 1 or Seq 2. (that would be a "0" for Seq 1 and a "1" for Seq2. 
midi_device.c already supports song select.

image-et-son, 2022-01-20:
Found out: there is already a wait mode. It is activated by holding From Tape down and pressing the seq and arp buttons. It starts the sequencer when a key is pressed. I have not checked MIDI real time event integration. The OS does something when receiving Clock Start and Clock Stop but I haven't really tried it out yet:

    case MIDI_CLOCK:
        ++synth.pendingExtClock;
        break;
    case MIDI_START:
        seq_resetCounter(0,0);
        seq_resetCounter(1,0);
        arp_resetCounter(0);
        clock_reset(); // always do a beat reset on MIDI START
        synth.pendingExtClock=0;
        break;
    case MIDI_STOP:
        seq_silence(0);
        seq_silence(1);
        break;

image-et-son, 2022-01-30:
Further enhancements for seq / arp start/stop via MIDI including wait modes etc. will be in the backlog --> separate issue

closed 2022-01-30 by image-et-son