image-et-son/p600fw #50: Add parameter sequencing

Colpocorto, 2021-12-28:
Some modern synths (both VA and real analog) have this feature.
Each sequence could store not only notes, but also parameter changes. 

One possible approach could be to support three simultaneous parameters per sequence that can be selected from a selection of "sequenceable" parameters.

Use cases:

-Different combinations of waveshape could be set ON or OFF at each step to simulate wavetable sequencing.
-On those models with the noise kit installed, some synth drum patterns could be implemented.
-Polymod amount could be changed dynamically to add some movement to patches.

Optional:
-Value changes between steps could be abrupt or smooth (selectable).

Variatons:

a) It could be added as an additional feature of the already implemented sequencer. Thus, parameter sequences would be global and not specific to a patch.
b) Sequences could be specific to patches and have a maximum 16-step length. This approach would need to upgrade the data structure and storage format of patches (including sysex support).

image-et-son, 2021-12-29:
Hi, thanks for posting this. It is not an easy decision. The downsides are: it makes a complex UI (double/simultaneous button pressing, different modes of operation etc.) even more complex. In fact that would probably be the most complex UI part. The little display can only help you so much. Also, this takes storage space and app code space, possibly too much. And then it might be a lot easier to control everything using MIDI CC. So the cost-benefit-evaluation doesn't look very good to me.

Note that you can already do quite a bit of parameter sequencing by syncing the LFO to the arp/seq. Maybe we can expand on that function?

Colpocorto, 2021-12-30:
I agree with you about the exponential complexity of the UI to add such a feature.
Syncing the LFO to the arp/seq could be indeed interesting.
Another approach to achieve similar effects is to define waveforms switches as a destination for LFO2. Thus, when the LFO waveshape reaches the top or bottom limits, the switch will be toggled or shifted.

An example of finite automata that could advance one step every time the LFO reaches a max or min value (this whole behaviour would be defined as a single LFO destination with 4 modes):

Waveseq mode 1 destination (OSC A&B are changed in parallel):

OSC A:  [SAW_ON,TRI_OFF,SQR_OFF] -> [SAW_OFF,TRI_ON,SQR_OFF] -> [SAW_OFF,TRI_OFF,SQR_ON] -> goto state 0
OSC B:  [SAW_ON,TRI_OFF,SQR_OFF] -> [SAW_OFF,TRI_ON,SQR_OFF] -> [SAW_OFF,TRI_OFF,SQR_ON] -> goto state 0

Waveseq mode 2 destination (OSC A changes, OSC B remains static):

OSC A:  [SAW_ON,TRI_OFF,SQR_OFF] -> [SAW_OFF,TRI_ON,SQR_OFF] -> [SAW_OFF,TRI_OFF,SQR_ON] -> goto state 0
OSC B:  [--------]

Waveseq mode 3 destination (OSC A&B are changed in parallel, values are inverted):

OSC A:  [SAW_OFF,TRI_ON,SQR_ON] -> [SAW_ON,TRI_OFF,SQR_ON] -> [SAW_ON,TRI_ON,SQR_OFF] -> goto state 0
OSC B:  [SAW_OFF,TRI_ON,SQR_ON] -> [SAW_ON,TRI_OFF,SQR_ON] -> [SAW_ON,TRI_ON,SQR_OFF] -> goto state 0

Waveseq mode 4 destination (OSC A changes, OSC B remains static, values are inverted):

OSC A:  [SAW_OFF,TRI_ON,SQR_ON] -> [SAW_ON,TRI_OFF,SQR_ON] -> [SAW_ON,TRI_ON,SQR_OFF] -> goto state 0
OSC B:  [--------]

matrix12x, 2021-12-30:
I like this idea and the new version you (@Colpocorto) propose is very easily implementable, in my opinion, as a modulation target for either of the LFOs. However, could this cause any audible pops since the oscillator waveform during a switch (from for example a square to a saw) may have a discontinuity? The VCO is an analog oscillator and when we choose to switch one of the waveforms on/off, it does not start from a zero phase. 

I guess this would be easy to try using MIDI CCs to hear what it sounds like. 

@Colpocorto would you mind mocking up an audio example?

Colpocorto, 2021-12-31:
@matrix12x I can't test anything because my Prophet-600 is not working right now (BTW, I'm Mastropiero at GearSpace).

You will have to wait for my 600 to be fixed or try to demo the switching effect by yourself. Maybe someone else could help here (el-folie?).

matrix12x, 2021-12-31:
I tried it out using CCs. very cool sounding. However, there were some discontinuities and you definitely heard them. Although, I don't think it was that bad. I could not figure out how to upload the wav file I recorded to this forum.

Colpocorto, 2022-01-01:
Yeah, the pops are quite audible. There is always a price to pay for having genuine VCOs ;-)

matrix12x, 2022-01-01:
You definitely see the discontinuities in the wav file. In the audio sample, I had the synth on unison, in hopes that the phase of one or more of the 6 VCOs would reduce this step. 

The "wave sequencing" does sound cool though.