image-et-son/p600fw #51: Bug: P600 remains silent on key presses after being left unused but powered on for a certain amount of time

el-folie, 2021-12-30:
The volume comes back by shortly moving the main volume knob.

image-et-son, 2021-12-30:
Hi, did you observe this in alpha 9? I did a change there and I just want to make sure, that I know where to look.

el-folie, 2021-12-30:
Haha, I was sure I was testing it on alpha 9 but you´re right - I´ll test again to verify!

matrix12x, 2021-12-30:
Alpha 9 I confirm volume drops to zero if I leave the P600 for a while. I think mine was on about 60 min before I checked it.

I have to fiddle vol knob for it to come back. See my code for fix.

el-folie, 2021-12-30:
I´m glad that Matrix12x could reproduce the bug - I couldn´t so far, crazy elusive it seems...

matrix12x, 2021-12-31:
at around line 1140 of synth.c we need to send the volume CV.

I never solved what causes the volume to degrade over time, and my fix was a brute force method.

image-et-son, 2021-12-31:
I think that the "cause" is quite simple: in the program you set the voltages using the DAC. But then the voltage degrades slowly due to natural leakage. Almost all voltages are set continuously/repeatedly, but the volume voltages is only set once a change is detected. I have moved the MVol  voltage update from continuous (cycle on in synth_update() ) to the conditional part. What needs to be done is to store the volume voltage internally and to update it continuously. For volume there are now 3 drivers to be consolidated: MVol knob, benderVolumeCV and MIDI CC.

BTW: there are 3 controls which are not stored in the patch and are therefore applied without using storage in currentpreset.[]: bender, mod wheel and MVol. Out of the three MVol is currently the only one without an internal storage variable.

matrix12x, 2022-01-02:
Alpha 10
I left on my P600 for an hour and did not have the volume drop bug. I will continue monitoring.

closed 2022-01-03 by image-et-son