image-et-son/p600fw #53: Bug: Spread value doesn´t return to 0 for attack time variation after once being used

el-folie, 2021-12-30:
Work around so far: Switching to Preset mode and back to Live mode.

Complete description with audio examples in the gearslutz thread.

el-folie, 2022-01-14:
It´s still like this on my P600: Switching from manual mode (with some Spread value) shortly to preset and back to panel mode, the Spread value gets lost and set to zero.

Does it really work on your P600s?

matrix12x, 2022-01-14:
I confirm I have the same exact issue as described by @el-folie

el-folie, 2022-01-14:
Thanks for confirming Matrix12x.

matrix12x, 2022-01-14:
@image-et-son you think the changes to storage.c and storage.h may have caused this? I have had that happen. if something is re-ordered or renamed.

image-et-son, 2022-01-15:
There is some very weird behaviour with transfer of menu parameter value to the live mode patch (to retrieve when switching to and from preset mode) and storage. The ext voltage and and the vintage values are not stored in either and reset to zero. Also when storing patches from live mode some menu parameters are reset to the last stored live mode patch, but only if they were the last value to be changed. This is a mystery and very important to resolve.

matrix12x, 2022-01-15:
I confirm the Ext Voltage parameter not being stored. In preset mode I modified a patch to add the Ext Voltage (noise on my P600) and I hit record to save the preset (for example in spot 01) then without hitting anything after the save, the Ext Voltage defaults back to 00.
Same thing happens if I switch to manual and then back to preset, or if I set the ext voltage while in manual and then switch to preset and back to manual, the Ext Voltage goes back to 00.

image-et-son, 2022-01-15:
Hi, sorry, the external CV was missing in 11 but I have it in my local code with exactly the same error or for spread/vintage. 

The hold pedal thing is a little crazy: since the parameter is in MIDI CC 64 it is in the right place so you can use the sp index directly for mapping to CC values (this type of design saves a lot of glue code :-), so is good). But for the patch storage the hold pedal is irrelevant. So instead of the hold pedal at that position the amp envelope speed is stored. And the sp for amp envelope speed (25?) later on is skipped in patch storage. I today added more commenting on this :-)

I have solved one problem: the menu parameter values form live more are only stored when you select another menu parameter, not when you change the value (which is OK) but when you store the patch from live live mode the code goes back to live mode an recovers the latest saved value (which may be different). So in the storage procedure I also store the latest live mode now.

  But the non-saved spread and ext volt remains obscure to me

image-et-son, 2022-01-15:
OK, that was a stupid mistake. I referred to *.steppedParameters[cpSpread] instead of *.continuousParameters[cpSpread] in the code. 

At least I found the error above in the processes.

matrix12x, 2022-01-15:
cool. and that makes sense for cpSpread.

el-folie, 2022-01-19:
Tested - works, values are retained when switching to preset and back top panel.

closed 2022-01-19 by image-et-son

matrix12x, 2022-01-20:
Alpha 12:
I confirm Ext Voltage(noise) and Vintage(spread) store correctly