image-et-son/p600fw #54: Speed not stored / recalled correctly after power cycle

image-et-son, 2021-12-31:


matrix12x, 2021-12-31:
This bug existed in my versions as well. Honestly I could not get it to recall speed with the patch. The bug probably exists in GliGli's original code.

Also, I don't think Arp/seq speed should be a patch level parameter. Because if it was, then when you change patches, arp/seq speed would change. So imagine you are playing a patch with the sequencer playing, and you go to audition different patches with that seq, if speed was a patch level parameter, then the speed would change.

On my unit on Alpha 9 (monad previous) speed always starts at zero on power on.

image-et-son, 2022-01-01:
Finally found it. There were some remnants of the speed parameter as part of the patch and the transfer of the parameter to a settings parameter was only half implemented.

closed 2022-01-05 by image-et-son