image-et-son/p600fw #56: Function to reset external pitch bend value

image-et-son, 2021-12-31:
Since internal and external pitch don't override each other but add to each other, any received external pitch bend can only be undone by external MIDI pitch bend. This should be reset at least when switching modes (preset and live) and preset number.

matrix12x, 2022-01-01:
How does the original SCI firmware handle this situation? 
I 100% agree that this should be reset at least when switching modes and preset number.

el-folie, 2022-01-01:
Just tested with Z80 in place: only a power cycle on/off resets the external pitchbend offset.

I highly regard it as a feature to be able to keep some serious down-tuning WHILE switching between sounds of live/panel and preset modes. A sudden change in tuning would disturb the experience. In my opinion if a PB reset is necessary it should be a seperate button combination like "Record + 3" or something. (Record + 3 was calibrating PB center on SCI OS)

image-et-son, 2022-01-01:
Thanks for checking this. I propose to incorporate it in the bender calibration function, e.g. when you calibrate the bender the external pitch bend is set to zero. Should this be when the user touches the bender or when entering calibration mode or when exiting it?

el-folie, 2022-01-01:
Great idea! 

I think when calibrating PB center by "From Tape + 33" and at this moment also to reset external PB values makes total sense as it´s like calibrating the internal tuning system. It´s related, so yeah, great idea!

image-et-son, 2022-01-01:
OK, I did such that it is triggered when entering PB calibration mode. That seems to most appropriate point.