image-et-son/p600fw #57: BUG: OSC B volume never goes fully to zero if Poly mod VCOB is assigned to filter

matrix12x, 2022-01-01:
OSC B volume never goes to zero

Settings: (in preset or manual mode)
All VCO A waveforms OFF
VCOA Volume max (99)
VCOB any waveform ON
VCOB volume zero (0)
Poly mod VCOB any value above half (or so)
Poly Mod Filter ON

you can still hear VCO B at a low level

(This does not happen with the polymod Env instead of VCOB)

I never noticed this before. It may be in prior Alphas. it may be hardware. I wonder if this is related to the sync bug?

image-et-son, 2022-01-01:
Hi, I played around with this for a while. I think that it is "normal". After all, the audio path from OSC B is not through the filter to the output but rather into the filter cut-off frequency. Although the audio input to the filer is silent, it is completely normal that the variation in cut-off frequency in the audio range can produce a (small) filter output. That depends strongly on the base cut-off frequency, the resonance and the filter chip characteristics - at least I got very uneven results from the different voices on my P600. From previous "experiments" I know that my voices 3, 4 and 6 have a higher resonance point, for example.

matrix12x, 2022-01-01:
Thanks. I had a feeling this was a "normal" hardware related thing.

image-et-son, 2022-01-01:
I'll close it. Nothing can be done here anyway.

closed 2022-01-01 by image-et-son