image-et-son/p600fw #61: Feature Request: add VCA as LFO modulation target

matrix12x, 2022-01-01:
New LFO target: VCA

see my old code

matrix12x, 2022-01-14:
Alpha 11, this works, but maybe needs to be "stronger" For example, I may suggest bit shifting the value to multiply it by 2x.

image-et-son, 2022-01-14:
Absolutely agree, and the same is true for the VCA target of the LFO. This is why I propose to introduce a strength parameter for VCA target for LFO (because of backward compatibility, otherwise I'd be happy to increase is flat). 

For vibrato I'll add a factor of 2 and we'll see if that is sufficient.

BTW: the vibrato contribution is already a factor of 2 stronger than the LFO, the function is:
ampLfoVal+=(synth.vibrato.output<<1)-(synth.vibrato.levelCV);

matrix12x, 2022-01-22:
Alpha 12 hot fix:
Vibrato to VCA sounds good. on my unit it can get down to silence, which I like.

image-et-son, 2022-01-22:
Thanks for testing. And yes, I have rewritten it such that it covers the entire value range :-) but I had to build in a cut-off so that additional variation does nothing if maxed out. I'll close this. If you find something odd or wrong we can open a new issue.

closed 2022-01-22 by image-et-son