image-et-son/p600fw #63: The action of the LFO amount pot should be smoother

image-et-son, 2022-01-07:
Idea would be to implement a curve like for the mod wheel but not introduce a patch parameter for the action curve. Instead old patch values should be mapped onto new ones.

el-folie, 2022-01-07:
Wow - you are reading my mind! For years I´ve thought the envelopes just didn´t sound correct. 

Maybe we could somehow transfer response curves from a CEM3310 - Prophet-5 - Repro 5 over to the P600 as a 5th setting? Also maybe the order of envelope shapes would be nicer as "Fast EXP - Slo EXP - Fast LIN - Slo LIN - CEM3310"

I also for a long time had the wish to be able to have the same Prophet-5 like "envelope thump/click" which is a lot like the "Attack-Hold bump" which makes a Minimoog Attack have such a great punch and impact. I personally would wish for a new envelope parameter "Attack Hold Time" that could be set from 0 to a few miliseconds. That way Minimoog sounds would be possible and with less hold time the Prophet 5 bump. If needed I can record audio exapmples to compare the bland but "scientifically correct" envelopes of P600 to Model D and Repro to demonstrate the tiny but huge differences in behaviour and sound.

image-et-son, 2022-01-13:
Thanks for testing this. I do have the feeling that the LFO amount impact on the different targets is not really balanced. In particular, the amount of VCA modulation is too weak compared to filter and frequency, e.g. in ranges where filter and frequency modulation is already wild, you cannot hear the VCA. Even though I don't like introducing menu parameters, I think it might be good to have a  VCA strength parameter, or - better - to separate it from the oscillator selection, and have a switch "off, min, med, max" (which of course would be one reason not to think about re-designating the former LFO shape switch for VCA target activation).

matrix12x, 2022-01-14:
Alpha 11 
I confirm, very smooth

el-folie, 2022-01-14:
I still hav a problem approaching high values for PWM to have a slow piercing modulation right at the brim of breakup to silence. I can´t fine-tune the LFO amount pot right to that point as it´s between 98 and 99 on my P600. To achieve that I need to add some mod wheel value of about 50. So maybe the LFO amount knob resolution approaching max at the top end could be a bit finer?

el-folie, 2022-01-14:
Of course I fiddled around with the OSC PW setting too (to find the exact middle position of the PW wave form) to get as close to peircing nasal sound with LFO depth at min/max LFO travel as possible. It´s rally just the last peircing bit missing on the LFO amount knob (between 980too less and 99= already zero/silence) for that special piercing satisfying sound.

So yeah, maybe one bit less of resolution may fall right on the piercing PWM spot, while hopefully still let LFO amount 99 be zero/PW silence at the edges...

Polarity reversal would be interesting for any mod target of course, maybe that´d be something for the menue?

image-et-son, 2022-01-15:
Action is too smooth now

el-folie, 2022-02-01:
To me action the knob action feels good.

closed 2022-02-01 by image-et-son