image-et-son/p600fw #64: The exponential decay and release curves of the envelopes are considered too short

image-et-son, 2022-01-07:
[envelope_decay.xlsx](https://github.com/image-et-son/p600fw/files/7828571/envelope_decay.xlsx)

image-et-son, 2022-01-07:
The open questions are
1) should bother, the decay and the release be modified? Currently these have the same shape when set to exponential
2) What is a good target shape, for example of a P5? It is likely not a pure exponential...

el-folie, 2022-01-07:
Wow - you´re reading my mind! For years I´ve thought the P600 envelopes didn´t sound correct yet.

Maybe it´d be possible to transfer the CEM3310 response of Prophet-5 - Repro-5 over to the P600?
And maybe a somewhat nicer order for envelope shapes would be "Slo-LIN - Fast-LIN - Slo-EXP - Fast-EXP - CEM3310"

In addition, a very important aspect of the CEM3310 - Prophet-5 envelopes is that little "attack hold time on max level" after the actual attack time has reached max level. This is part of what makes the Prophet-5rev3 have that thump/click/acoustic response. The same happens on a Minimoog, the max level at attack is being held even longer and that makes for that unique punch and impact in Minimoog sounds. So, a new "Attack Hold Time" parameter with control from 0 to a few miliseconds would bring great benefit to the envelope sound.

matrix12x, 2022-01-07:
I would like to note that the issue some people may be bothered by is the "abrupt" ending of the envelope on some peoples P600s that Gligli introduced because some peoples P600s VCA did not shut off all the way. They may be perceiving this as too short of an envelope. There was some discussion of this in the GIT issues on GliGli's branch and I seem to remember in the earlier pages of the gearspace.com thread.

https://github.com/gligli/p600fw/issues/90

el-folie, 2022-01-07:
> I would like to note that the issue some people may be bothered by is the "abrupt" ending of the envelope on some peoples P600s that Gligli introduced because some peoples P600s VCA did not shut off all the way. They may be perceiving this as too short of an envelope. There was some discussion of this in the GIT issues on GliGli's branch and I seem to remember in the earlier pages of the gearspace.com thread.
> 
> [gligli#90](https://github.com/gligli/p600fw/issues/90)

That was def the case but wasn´t it solved long ago and wasn´t it also dependent on hardware/CEM3372 rev.? 

At least in the Starsky Carr video he just mentions the abrupt silence at the end of the LINEAR envelopes as being bad compared to the rev4. But that silence is the same on a non-gliglified P600, it was always there with the linear envelopes. The new exponential slo/fast envelopes don´t end with sudden silence, which is good, but they are still too short in their length/A-D-R times and their shape is far away from usual Prophet-5 behaviour.

matrix12x, 2022-01-07:
As you said, I think it is hardware dependent. it was solved for some people and caused for others. In August 2021 (or so) someone posted on GLIGLI's GitHub that they have the issue still. 

Of note the data sheet says "the point at which the VCA in the CEM3372 shuts off is very well defined (+100mV ± 50mV)."

I usually use the Expo Fast for plucks and basses, so it works well for me how short it is. But outside of that use case I do get what you are saying. A few mS hold on the attack would add a nice "zap."

image-et-son, 2022-01-07:
Hi, implementing a different release curve is trivial once I have the data. I did some research, but there is a lot out there and nothing really "canonical". For the exponential shape the release time effectively changes the decay rate but it's the same curve shape. So any real difference would be a change in the actual curve, e.g. something like a linear phase followed be a smooth exponential curve. Do you have any information - if only anecdotal - on what curves are used in the P5 and like? 

As far as the offset correction is concerned: if the code supresses a run-off offset then this would lead to sudden stops. I think I have had something like this and I find sudden silence irritating. I haven't found anything in the code producing an offset in the code, though..!?

matrix12x, 2022-01-07:
Line 40 of synth.c

// The P600 VCA completely closes before the CV reaches 0, this accounts for it
#define VCA_DEADBAND 768

then later at line 953:
 if(va)
            va+=VCA_DEADBAND;

el-folie, 2022-01-07:
I also tried to find some precise CEM3310 data or Prophet 5 envelope shapes, but nothing significant so far. If it`d help I could do recordings of the Repro-5 behaviour with any envelope settings you may deem necessary to find a mathematical approximation. The Repro shapes should be close enough to a real Prophet 5.

As a first simple observation, the P600 decay/release shape is falling too steep too early in comparison to the Prophet-5, which much longer holds and then much more slowly decays from maximum decay level. Also, at the end of decay/release the Prophet-5 seems to extremely slow down before reaching zero level, like a discharging capacitor. But all of this is just superficial reflection. So, would audio recordings/comparisons help you in any way?

image-et-son, 2022-01-07:
> Line 40 of synth.c
> 
> // The P600 VCA completely closes before the CV reaches 0, this accounts for it #define VCA_DEADBAND 768
> 
> then later at line 953: if(va) va+=VCA_DEADBAND;

OK, I was even fiddling around close to this. This is a VERY crude thing :-( But for those who don't have a problem with the amplitude suddenly going to zero, this will create a constant small offset. I even wonder what happens if the incoming va is already UNIT16_MAX?

el-folie, 2022-01-07:
https://www.eddybergman.com/2019/11/synthesizer-build-part-3-envelope.html

If I interpret the 2nd scope shot correctly, where he varied the decay time, the "AS3310" even seems to alter its shape depending on decay time, becoming more linear? Does that make any sense?

matrix12x, 2022-01-07:
It would not be because of the AS3310, if it did change shape, it would be because of the ADSR circuit.

el-folie, 2022-01-07:
Thx Matrix12x! I never built an envelope circuit...

https://modwiggler.com/forum/viewtopic.php?t=83193&sid=978324f3f55adc48bf26603980057799

In that thread user oldcrow presents a CEM3310 equivalent circuit and on page 2 an Italian user shows the correspondent mathematical equations. Maybe those could be of some real help?

image-et-son, 2022-01-07:
[image]

I did this exact graphical representation of the current p600 shapes for the documentation. It goes into "saturation" very quickly as you can see in the decay phase. Still, you can compensate it easily by using longer times. I think subjetive difference is about the shape, presumably more linear at the start. I'll have a look at the links.

el-folie, 2022-01-07:
Yes, 3310 looks and sounds like being more linear at the start and much more exponential/sloooooower at the end.

Graph looks ace!

matrix12x, 2022-01-07:
@image-et-son I think we should try the proposal you have above in the .XLS and see how it sounds. Maybe initially introduce it as 2 more env options.

el-folie, 2022-01-07:
[CEM3310_datasheet_long.pdf](https://github.com/image-et-son/p600fw/files/7831072/CEM3310_datasheet_long.pdf)

Just found this longer CEM3310 pdf which includes the exact equations of the envelope stages.

matrix12x, 2022-01-07:
Of note, in the formulas in that datasheet, VT is thermal voltage (kT/q). I.e. k is Boltzman's constant. At room temp this gives us a VT of about 25.85 millivolts. (depending on where you live)

matrix12x, 2022-01-08:
This webpage has an interesting video where they change the shapes of the ADSR segments:
https://www.earlevel.com/main/2013/06/22/envelope-generators-adsr-video/

matrix12x, 2022-01-08:
I put the formula for decay for the CEM3310 from the data sheet  into the XLS above and plotted it over the other two plots. That seems to explain what people are hearing.
[image]

el-folie, 2022-01-08:
Wow - those plots look absolutely amazing! If I´m not mistaken the blue plot could be the P600 linear envelope. The yellow one looks like I expected it, much smoother slope than the GliGli EXP shape.

This new development and the prospect of maybe having real CEM3310-like response on the p600 in the future is totally exciting to me. This could easily be a game changer for the p600 sound and behaviour. :)

image-et-son, 2022-01-08:
Hi, nice work. However, it is just the same curve but with a different scale on the time axis. In software generated envelopes you need to stop somewhere, while the pure exponential never strictly goes to zero. Therefore you define a cutoff point and then scale the curve so it goes to zero exactly at that point (and you want the point well into the near zero decay phase of the curve). If I do this to the CEM3310 function and then put in a scale factor of about 1.77 I get EXACTLY the curve in the GliGli Lookup. The only difference is the scale you have available for the release time parameter. 

 So I see 3 conclusions for me on this: 1) I am currently not really sure what Starsky meant exactly and it in my opinion could still be the drop from the quick-and-dirty offset correction. We best ask Starsky! 2) The cutoff point in the GliGli curve is so far out, that while the decay and release periods in slow envelopes are up to 50 seconds long, the max effective decay time is super short, only about 5 seconds, which seems too short, esp. compared to the linear shape, e.g. a factor of 5 to reach half max amplitude. 3) There is talk about a more linear decays variation in rev 1/2. Obviously that would be another shape compared to CEM3310.

el-folie, 2022-01-08:
2.) So, if I understand you correctly, just altering the scale factor of the EXP shape leads to correct CEM3310 shape? In that case, and maybe also to be able to approximate rev1/2 shapes (SSM envelopes), maybe we only need an envelope scaling factor paramater that could be user set to taste? Would bring the greatest flexibility.

1.) Starsky disliked two aspects: the too steep EXP slope on the p600, as in the too fast decaying decay phase. And second, the sudden drop into silence in LIN modes.

image-et-son, 2022-01-08:
I don't think that scaling would really change anything, One aspect of exponential functions is that the decay steepness is determined only by the start point and the decay time. All exponentials are the same. The only potential difference of scaling would be that some users may want decay times to be longer than what can be achieved by turning fully up to 10 on the P600.  But it will not change the characteristics of the decay.

I looked at Starsky again (I don't want suggest that his opinion is the ultimate reference point, but in the end it was his main criticism) and he said, the sound rolled off too fast, then turned up up the release time and concluded that this wasn't working because then the overall decay time is too long. He then described the P5 roll off as holding a bit before decaying. In my interpretation this is a different shape. Here is his screen shot:

[image]

What you can see at the bottom is that the relative durations of linear and exponential are too strong. But the real difference is that the P5 shape is more linear before going into smooth decay. So my proposal would be to try to introduce that kind of shape!

el-folie, 2022-01-08:
Thank you for explaining, clearly I misunderstood.

Yes, introducing the exact CEM3310 shape would be a dream.

(And as mentioned in my last post, Prophet-5 release phase always seems to be just a smidge smoother/more linear and longer in its release til reaching zero than the decay phase.)

image-et-son, 2022-01-08:
OK, we have a plan :-)

el-folie, 2022-01-08:
But one important aspect here: we don´t know which envelope mode Starsky was using when plotting the decay graphs. We don´t know if those graphs show rev1/2 or 3 envelopes. Personally I´d think the graph shows rev1/2 as in the video he is in rev1/2 mode of the filter which also switches to rev1/2 envelopes if I remember correctly.

el-folie, 2022-01-08:
In Starskyý graph, left and right Prophet 5 decay phase examples are identical. I used photoshop to extract the right shape, changed its colour and copied it to the left to compare both shapes. In my opinion this could be rev1/2 SSM 2050 shape:
[image]

matrix12x, 2022-01-08:
@el-folie I agree. this seems like the "more linear" Rev1/2 env shape. The SSM2050 data sheet shows a non-linear curve, but I don't see any formulas or details on the shape of the curves.  

https://www.ka-electronics.com/images/SSM/SSM2050.pdf

If we really wanted to overcomplicate this, I don't mind buying the CEM chip, building the circuit, and sampling the output to use as a table. I can do the same for the SSM2050. However: 1) I need to find/buy these parts, 2)maybe this is severely overcomplicating just trying a number of different slopes and hearing what sounds/feels best.

el-folie, 2022-01-08:
Thank you for confirming my thoughts.

And also a very kind offer to analyze the correspondent chip envelopes but honestly I assume that would be too much of a hassle for you to do, and I have a feeling that imogen already has figured out a simpler plan.

image-et-son, 2022-01-08:
It depends not only on the chip but also on the surrounding circuitry, e.g. how RxCx is provided. We do have a qualitative shape and I think we can work from there. Would be nice to have someone on board with a P5, though. In any case, I have posted a question on youtube (although I seem to have problem with my user name there, which isn't shown :-( ). I could not find official information on whether and how the rev 1/2 vs. rev 3 setting should influence the envelope shape. It is meant to change the filter characteristics.

el-folie, 2022-01-08:
I will do a few tests with Repro-5. U-he emulations are usually top notch will be close enough to real Prophet-5 envelope behaviour...

el-folie, 2022-01-08:
> It depends not only on the chip but also on the surrounding circuitry, e.g. how RxCx is provided. We do have a qualitative shape and I think we can work from there. Would be nice to have someone on [...]

It´s here in the 1st parapraph from Sequential themselves:
https://www.sequential.com/product/prophet-5/

So, the rev switch really alters the envelope shapes.

image-et-son, 2022-01-08:
OK, so I was not looking very thoroughly :-\

el-folie, 2022-01-08:
Okay, recorded some audio of P600 and Repro5 to compare. Did screenshots and put them together for visual comparison.

Results for quite long decay times:
- It´s impossible to match p600 and Repro envelopes.
- p600 needs more decay time to decay a bit more linear but will then sound too long and doesn´t sound as nice as Repro when approaching zero, Repro does it more slowly/natural.
- p600 has a rapidly falling steep decay slope at the attack in comparison to Repro
- Repro sounds more linear at the start of the decay phase and extremely exponential at the end.

Two graphs:
1. Comparison of Repro Decay & Release phases at the same value of 80. It clearly showed that release is always longer despite the same value of 80. This is true for ideal mode and analog mode, which models a Prophet 5 with CEM3310 chips.
2. Comparison of p600 and Repro decay slope. They were impossible to match concerning levels, times and behaviour. But the graph shows that the p600 decay is more exponential, steeper, falling faster, while Repro more linear, more gradual at first and sweeter, slower to vanish into nothingness at the end approaching zero.

1:
[image]

2:
[image]

Hope this is of help somehow.

el-folie, 2022-01-08:
Here I copied the Repro decay slope on the p600 slope for comparison. Even without level matching one can see that the p600 has a much more exponential slope.

[image]

el-folie, 2022-01-08:
I went through it all again to get perfect timing and level. 

Attached is the stereo audio of "left Repro - right P600". In the end P600 cuts off while Repro vanishes exponentially. The audio file needed to be zipped to be able to upload it to github.

In the graph both signal peaking levels were the same and also almost the same length. P600 decay value 83 was too short, 85 too long. So this is the most precise comparison possible at this given decay length. The difference in shape is visible.

[L_repro5_decay80 - R_ p600_EXPdecay84.zip](https://github.com/image-et-son/p600fw/files/7833820/L_repro5_decay80.-.R_.p600_EXPdecay84.zip)

[image]

image-et-son, 2022-01-08:
> It´s here in the 1st parapraph from Sequential themselves: https://www.sequential.com/product/prophet-5/
>
> So, the rev switch really alters the envelope shapes.

You can see in the video, that switch is on rev 1/2. But to my understanding it changes the filter envelopes and the discussion was about amplitude. It is probably not worth digging deeper there and instead we just try out a different shape and listen. Maybe we can even get an opinion from Starsky, too. A least he seemed interested and replied to my question. Note: I wanted to ask him about  rev 1/2 vs. rev 3 but it seems like a silly question to me now.

el-folie, 2022-01-08:
I don´t think it´s a silly question at all as rev1/2 and rev3 envelopes really are different. The rev switch alters both envelopes for amp and filter, which are the same for each, so the only difference will be the slopes of rev1/2 and rev3. Maybe Starsky would like to contribute a graph of rev1 and rev3 slopes, would be nice of course. 

As for rev3 only I think the above Repro slopes are quite telling. If you also need attack shapes I can do those too...

image-et-son, 2022-01-08:
Concerning your L/R recording. Good job! 

You pretty much matched the main decay phase (very slightly higher on the P5 side) but in the last 3rd it's completely P600. So when you make them the same in terms of duration of the release (that is, for example, time to half amplitude) then the P600 rings on too long. So in essence, we need to find a curve that goes down faster than the exponential in the middle and then tails off flat. 

Do you see it the same way? I find it difficult to reconcile the different descriptions but this is my take on the situation at the moment.

matrix12x, 2022-01-09:
Watching the video that you guys are referencing, at around 2 min or so, it's not the Exp envelopes that seem to be the issue, it's the unit he has cuts out at the end of the envelope. If you look at the image below it's the abrupt ending of the note at the end of the release. (the P5 seems to have a nicer fade for the last moments of the release. I think this is the VCA_Deadband and the CEM3372's VCA shutoff at (+100mV ± 50mV).

Then at the 3 min mark he is comparing the P600 Slow Exp to the mostly linear P5 V1/2, mostly because he seems to like the fade better at the end of the P600 slow exp than the P600 slow linear, but clearly this P600 env shape would not match the mostly linear P5 V1/V2 (SSM), which he comments that it does not.

I think the VCA_deadband/abrupt ending for the linear should be definitely fixed. 

Here is the P600 (linear) in his video and then the same note on the P5  (Ver1/2 mode) sampled at around the 2 min 18 second mark:
[image]

el-folie, 2022-01-09:
> Concerning your L/R recording. Good job!
>
> You pretty much matched the main decay phase (very slightly higher on the P5 side) but in the last 3rd it's completely P600. So when you make them the same [...]

Actually, if you listen closely to the audio file in the last third at high volume, it´s the P5 that rings out longer while the p600 just suddenly stops from being louder. That´s why the P5 sounds more airy and reverb-like in its last moments of the decay and release phases.

I´ll try to use your own words to describe a possible shape close to P5:
(- at the beginning there´s the "max level hold time", maybe about 2ms thick impulse on keypress. Issue #65)
- this is followed by an almost linear but slightly exponential decay slope
- this only slightly exponential slope first decays slower and when reaching saturationthen decays faster, in comparison to the p600 EXP env 
- after "saturation" of the curve, the P5 tails off more slowly and gradually than the p600, which suddenly cuts off

You are right, hard to describe with words ;-) But I do have great respect and hope for your abilities and taste in these matters and I´m sure, by testing out some shapes we will eventually find the right amount of decay slope and release behaviour.

el-folie, 2022-01-09:
@matrix12x

Gligli might have left the LIN env modes with the abrupt ending because actually that´s how the original Sequential OS envelopes behaved. But if you ask me, I too would rather like a smooth ending on any envelope shape. It just always sounds more natural and musical.

image-et-son, 2022-01-09:
The graphs are superb and instructive. I believe all observations are consistent, at least I hope. I have put together a representation of what I propose to do based on the P5 decay curve in the post from Matrix12x:

[image]

The P5 envelope is linear in the main decay phase with a flat roll-out: I propose to make a curve with a linear part (yellow line) and a matched exponential (orange line) for roll-out. You can see, that depending on the "saturation" point where the two are joined the matched exponential will be fast but smooth. When you try this behaviour on the P600 the linear shape doesn't roll off nicely (plus the DEADBAND problem, which needs to be solved) while the exponential dies too quickly or - if you try to compensate this by longer decay time - it rolls off too long. These are the two red dotted exponential curves (the short one tries to match the point to zero amplitude and the long curve tries to match the main decay slope).

el-folie, 2022-01-09:
Spot-on, imogen!

The only thing to keep in mind is, though, that the graph shows the more linear SSM2050 behaviour. So CEM3310 will be a second endeavor and slightly exponential compared to the yellow line.

image-et-son, 2022-01-09:
Hi, that' the new curve. The way I created it was heuristic. But in essence it is a linear/exp mix down down about 10% after which it goes into a matched exponential. There are two main parameters: the mix (linear vs. slightly curved) and the match point (longer vs. shorter roll off). We can have several iterations if there is room for improvement.

[image]

matrix12x, 2022-01-09:
@image-et-son I agree 100%. I think your proposal is a great path forward.

"BTW, what´s the name of the audio editor/program you used to do the graph?" 
@el-folie I used Audacity (Mac). It is free and really useful.

I am really enjoying this discussion. :-)

matrix12x, 2022-01-09:
@image-et-son if you spit me out the table for your proposal above, I can put it in ADSR_lookups and compile asap (and drop the .syx here for @el-folie) to try it out.

el-folie, 2022-01-09:
I tried to match Starsky´s decay time example with Repro5, so a P5rev3 CEM3310 env behaviour.
Both audio and graph supplied, maybe it helps. The audio actually goes on until the end of the graph, that´s how extremely exponential the decay decends to zero on the CEM3310 envelope.

[Repro5_CEM3310_tweak analog_decay-value69.zip](https://github.com/image-et-son/p600fw/files/7835195/Repro5_CEM3310_tweak.analog_decay-value69.zip)

[image]

image-et-son, 2022-01-09:
> @image-et-son if you spit me out the table for your proposal above, I can put it in ADSR_lookups and compile asap (and drop the .syx here for @el-folie) to try it out.

[hybridcurve_lookup_v1.txt](https://github.com/image-et-son/p600fw/files/7835218/hybridcurve_lookup_v1.txt)

There is still some smoothing to do at the very end, but I think for a first shot it's good enough!

image-et-son, 2022-01-09:
> The same graph as above, just with a shorter time scale to better see if there´s maybe also a linear portion at the beginning. But to my eyes it´s a completely exponential shape from beginning to end for CEM3310: 

To me also. I can check by matching it. But pretty sure it hasn't got that obvious linear part.

el-folie, 2022-01-09:
Haha, it feels like team work in a science lab, it´s great! :)

matrix12x, 2022-01-09:
@el-folie  To me this graph looks identical to the P600 Exponential curve. (which is good)
Per @image-et-son 's comments way above (when I added the function for the CEM3310), when it is scaled correctly, it matches the P600's Expo curve exactly. 

Also, here is the .syx using the new table to replace the old table just for testing purposes (I didn't get to test yet):
[p600firmware.syx.zip](https://github.com/image-et-son/p600fw/files/7835237/p600firmware.syx.zip)

image-et-son, 2022-01-09:
Yes

el-folie, 2022-01-09:
Thank you for providing the test compile for SSM first test.

Using the new EXP SLO mode I tried to get as close as possible to Starsky´s b-note decay comparison in his video at 02:20.

Audio attached, I think is quite close with the Prophet still decaying somewhat slower/smoother.
[Starsky_rev1_decay_match_p600-SSM1_test.zip](https://github.com/image-et-son/p600fw/files/7835277/Starsky_rev1_decay_match_p600-SSM1_test.zip)

matrix12x, 2022-01-09:
Here is the graph using Audacity of the note you played above:
[image]

el-folie, 2022-01-09:
A more musical audio example, it´s nice how the SSM style linear decay makes for a thicker sound before the release. For demonstration the release phase at the end of audio.
[SSM1_test_chords_p600firmware.zip](https://github.com/image-et-son/p600fw/files/7835335/SSM1_test_chords_p600firmware.zip)

matrix12x, 2022-01-09:
I have also been testing it and I really like it. :-) Great work guys.

This "SSM" shape is the envelope that "linear" (P600) should have been. @el-folie good examples.

matrix12x, 2022-01-09:
@image-et-son so, assuming this table is cool, I assume we will be adding 2 new env shape options SSM Fast and SSM Slow, instead of replacing linear.

I guess the next steps are modifying ADSR_lookup to have 2 tables (EXP and SSM), update UI.c to include the fast and slow SSM, and tweak synth.c to incorporate this.

image-et-son, 2022-01-09:
> Thank you for providing the test compile for SSM first test.
> 
> Using the new EXP SLO mode I tried to get as close as possible to Starsky´s b-note decay comparison in his video at 02:20.
> 
> Audio attached, I think it is quite close, with the Prophet5rev4 SSM at the end still decaying somewhat slower/smoother to zero. [Starsky_rev1_decay_match_p600-SSM1_test.zip](https://github.com/image-et-son/p600fw/files/7835277/Starsky_rev1_decay_match_p600-SSM1_test.zip)

I will create one more variant to make it smoother.

el-folie, 2022-01-09:
@image-et-son

Great - looking forward to it. 👍

[50 more replies]

closed 2022-02-01 by image-et-son