image-et-son/p600fw #65: Feature Request: New parameter to "hold" attack for about 5 or 10 mS before starting decay

matrix12x, 2022-01-09:
Wow - you´re reading my mind! For years I´ve thought the P600 envelopes didn´t sound correct yet.

Maybe it´d be possible to transfer the CEM3310 response of Prophet-5 - Repro-5 over to the P600?
And maybe a somewhat nicer order for envelope shapes would be "Slo-LIN - Fast-LIN - Slo-EXP - Fast-EXP - CEM3310"

In addition, a very important aspect of the CEM3310 - Prophet-5 envelopes is that little "attack hold time on max level" after the actual attack time has reached max level. This is part of what makes the Prophet-5rev3 have that thump/click/acoustic response. The same happens on a Minimoog, the max level at attack is being held even longer and that makes for that unique punch and impact in Minimoog sounds. So, a new "Attack Hold Time" parameter with control from 0 to a few miliseconds would bring great benefit to the envelope sound.

Edit: 
What I´m talking about from listening could also just be some kind of "decay hold" due to an overshoot and limiting of the control voltage at the decay stage. Whatever it is it makes for that bump/punch and is one significant reason why digital envelopes almost never reach the impact of fast analogue envelopes.

Edit2:
For example - to achieve a similar attack on the P600 I need to reduce Amp Env Sustain to about 50-70 % and set Amp Env Decay to about 1.2. On a Prophet-5 - Repro-5 this little bump is built into the Amp envelope, even for simple "only amp sustain up" sounds. Adding the filter env then adds even more thump/click. With perfect digital envelopes this effect is not there unless it´s being taken into account in envelope designing/coding.

_Originally posted by @el-folie in https://github.com/image-et-son/p600fw/issues/64#issuecomment-1007381491_

el-folie, 2022-01-09:
Yeah, maybe this should be a seperate topic from the envelope shape discussions. Thanks!

matrix12x, 2022-01-09:
Also, what is the limitation on minimum attack time? Is it hardware (I.e. a minimum response time of the CEM3372)? or is it time for the code to execute?

el-folie, 2022-01-09:
That´s also a good question. The p600 with and without gligli OS def smears the attack a bit. There are two built-in lag circuits to smooth out the CV S/H steps for "filter CV in" and "final VCA CV in" at the CEM3372s.

But whatever lag those circuits may introdude, a seperate "max level hold time" would start after the smoothing circuit lag time, so that the more substantial impact of, say, a Moog envelope hold time, would still be retained and perceivable.

I did some tests a while ago to see how much "delay" the lag circuits introduce. For example, just using the filter oscillating sine with key tracking on full and a slow highest/lowest note sequence playing in 1-note unison. This test setup clearly made the lag perceivable as a short "pitch glide effect". From that point on I never wondered anymore why the P600 doesn´t have the extremely fast and tight impact of (some) analogue envelopes, like an SH-101 or JP-4 for example. And I guess the lag circuit for the amp envelope CV will do something similar, only harder to detect by ear. Of course there´s the question, if we still need those lag circuits at all with the mcuh faster update rates of the Teensy... I just didn´t dare so far to test this by soldering out the lag parts... ;-)

matrix12x, 2022-01-09:
I wonder if we could easily model playing with those R and C values to change the time constant? (I no longer have any SPICE modeling apps)
However, I honestly don't mind playing with the hardware on mine if we make some educated guesses. I have to review the service manual and the CEM3372 data sheet to see if there are any issues. I may sample the CV using a scope before and after the lag circuit. well, maybe before the buffer before the lag circuit.

[image]

matrix12x, 2022-01-09:
Check out October 3, 16, and Feb 28 of this blog:
https://prophet600.blogspot.com/2013/02/teensy-p600-alpha-installed.html

Also, interesting video debunking min attack time being slow as a result of envelope performance for the P600. 
https://www.youtube.com/watch?v=vo0_CYZt414

el-folie, 2022-01-09:
Thanks!

Re. the video, question still is if he probed the amp CV before or after the lag circuit. But if probed before, there´s no need to change anything parts-wise for the amp env CV. The filter CV definitely has lag though and could be a matter of discussion.

matrix12x, 2022-01-09:
if you compare the filter lag circuit to the VCA one, the VCA lag circuit kicks in a 10K resistor in parallel with the 100K one once the voltage goes above 2.4V (which is when those specific diodes conduct) and the cutoff freq goes from about 15.92 kHz to about 159.2 kHz. The filter one seems to always be at 15.92 kHz. 
design wise, I'm assuming the zipper noise/stepping is more offensive when you hear it in the filter. I wish the filter lag circuit had the 10K in parallel. 

Although you have a good point. these lag circuits were designed around a 4 bit envelope and an update of 5ms, which was super steppy. Maybe they are not needed. Again, I don't mind trying out some hardware mods. If I did, I would initially short R445 on voice 1 to see how that sounds. a second experiment might be to lift one end of D401, D402 and put a 10K resistor in series.

el-folie, 2022-01-10:
@matrix12x 

Of course we´re derailing the "attack hold time" topic with the lag discussion, but it´s too very interesting.

For demonstration purposes of the filter CV lag circuit I did this audio file:

[p600_filterCV_lag-demo_C0-C5-seq_unisonON-OFF.zip](https://github.com/image-et-son/p600fw/files/7836237/p600_filterCV_lag-demo_C0-C5-seq_unisonON-OFF.zip)

It´s just a simple filter sine with key tracking, tuned to taste so that both C0 and C5 are perceivable, running in a slow C0-C5 alternating sequence. Then just switching unison on and off a few times. 

This is what I assume I hear:
When switching unison on, one can clearly hear the short lag of the filter CV when trying to reach the high or low note, as in unison it´s always the same single filter, whereas in poly mode the high and low notes are being reassigned to individual filters, so that their pitch always stays the same. One can also hear how switching back to poly mode takes 6 "note ons" for the 6 filters to get back to a steady pitch without the lag circuit´s short pitch glide effect.

So, if I´m not mistaken, exactly this pitch glide effect makes it impossible for the p600 to let the individual voice filters have a fast enough CV changing rate to sound really snappy? If so, a hardware mod could maybe help. Of course a jump of 5 octaves in pitch is quite big so that the pitch glide gets very perceivable. At typical smaller steps like one octave it´s barely noticeable, but still there of course. 

As always, I´m also very sceptical about any findings:
Or do I hear something different here? Maybe S/H capacitor charging time, though I´d guess this to be a multitude faster than the lag circuit. Or maybe there is a CEM3372 filter CV input lag on the chip itself, like you said, a chip response time? Whatever it is, this is one reason for non-snappiness in mono mode, unison and poly mode use cases where 6-voice-filter-chords are alternated on high and low areas of the keyboard.

matrix12x, 2022-01-10:
RE: switching unison on, instead you can turn off voices 2-6, and you can then test the single voice. This way it takes more things out of the equation.

Also, good point. re digressing. On that note...

I think we can implement the hold similar to the LFO/vibrato delay. We may need some logic, so that if there is no hold value I.e. zero that it would just skip the hold code.

in ADSR.c change line 128:
LOWERCODESIZE void adsr_setCVs(struct adsr_s * adsr, uint16_t atk, uint16_t ath, uint16_t dec, uint16_t sus, uint16_t rls, uint16_t lvl, uint8_t mask) // adding the ath as a unit16_t

 // also add this, but need to change the mask number, I'm not sure how the mask here works, but I see every ADSR segment has a diff mask number....
	if(mask&0x01 && adsr->attackCV!=ath)
	{
		m=1;
		adsr->holdCV=ath;
	}

I think synth.c lines 610 and 641 of the current code would need to be changed:
(line 610) uint16_t aa,ah,ad,ar,fa,fd,fr; //I think this needs a "ah" added for the hold value
(new line) ah=currentPreset.continuousParameters[cpAmpHold]; //new line that needs to be added

(line 641) adsr_setCVs(&synth.ampEnvs[i],aa,ah,ad,as,ar,0,0x0f);  // add "ah" here too

UI.c needs a new parameter added at line 42:
/*2*/ {.type=ptCont,.number=cpEnvHold,.name="Hold"},

adding the usuals to storage.c and storage.h

el-folie, 2022-01-10:
@matrix12x 

Testing the above sequence with turning off all other voices, so that only one voice would sound, led to the same result. The lag of the above audio demo is constantly there with only one voice/filter playing, switching from to unison doesn´t alter it. So the lag is def a constant part of each individual filter CV.

matrix12x, 2022-01-16:
The top image shows the attack at zero for the P600 (top) vs a Sh-101 (bottom):

[image]

el-folie, 2022-01-16:
So I guess the graph shows the reaction time of the amp CV smoothing filter by the slightly smaller first sawtooth on the P600?

matrix12x, 2022-01-16:
I removed C462 (the cap for VCO 1 only):

[image]

Here is VCO 2 (still has lag cap):
[image]

Removing the cap definitely makes a difference. I'm going to see what other issues may have been caused by removing this cap. Any suggestions for how to test?

el-folie, 2022-01-16:
That´s some significant news! The CEM3372 data sheet also says that those "additional hold caps" are only needed in case of using the internal chip buffers for the S/H CVs. But the P600 already has external S/H CV buffers. So maybe we could really get rid of those lag/hold circuits to get a more direct sound at pin 15 (filter cv in) and 13 (final vca cv in).

matrix12x, 2022-01-16:
@el-folie I'm going to pull out C447 (filter lag) how should I test this?

el-folie, 2022-01-17:
Exciting! I don´t know if that´s enough though, as the rest of the lag circuit might still have an influence on the CV. 

Generally, a superb test for filter CV reaction speed would be to just use a single voice in mono-unison, oscillators off, filter-sine on with key tracking 100%, and then sequentially (lol!) pressing C0 and C5. If you don´t hear any filter glide/slide/portamento effect anymore like in my above audio example of the two note simple up/down sequence - then the CV is directly coupled to the CEM3372 at the best speed currently available (without further mods).

el-folie, 2022-01-17:
> I removed C462 (the cap for VCO 1 only):
> 
> [image]
> 
> Here is VCO 2 (still has lag cap): [image]
> 
> Removing the cap definitely makes a difference. I'm going to see what other issues may have been caused by removing this cap. Any suggestions for how to test?
> 
> There is a definite difference without this cap for the better so far.

I´d guess without that cap on voice 1 the VCA envelope response might sound more direct/faster/thicker, maybe even clicky in some cases? But I´d like that more than a tamed down attack.

matrix12x, 2022-01-18:
Yes, without the cap on the VCA its definitely faster.

Ok, VCF lag cap (C447) on voice 1 removed and I do not see or hear any lag:

[image]
[no lag cap voice 1.wav.zip](https://github.com/image-et-son/p600fw/files/7892895/no.lag.cap.voice.1.wav.zip)

el-folie, 2022-01-19:
And the audio file was done in one-voice-unison?

If yes, we´re getting a new synthesizer when removing the lag caps...

I´d be freaking out!

matrix12x, 2022-01-19:
Correct. one-voice-unison.

el-folie, 2022-01-19:
That´s really huge news then! I´ll check that out on my next analog board removal by also first removing just the lag caps on voice 1 to hear the difference within a given sound. I imagine it to be like night and day for some sounds with a sharp attack...

matrix12x, 2022-01-19:
For voice 1 you need to get the keyboard out of the way to access these two caps. But other then that they are pretty accessible. I just clipped the leads from the top of the PCB.

I will be doing this mod for the remaining voices. 

All of this being said, I still want a 4-10mS hold parameter for the ADSR. (I.e., AHDSR)

el-folie, 2022-01-19:
Hahaha... oh my... I went all out and soldered them all off the board. Then did some tests... Audio attached.

Results:
0. Without the caps the envelopes are a few ms faster in attack and finally sound analogue. The lagcaps cut away exactly the first few ms that make for that analogue goodness impact that we love in old hardware envelopes. Without the caps I needed to set the Attack knob to about 6 (EXP-SLO shape) to simulate the fastest attack time with the lagcaps. Bad! The very first cutting attack hit is def cut away by the caps.

1. Bad news: the caps are necessary though, unfortuately even still with the Teensy, as the digital noise is perceivable on "note ons" and note releases, in every A-D-S-R movement and in their phases audible stepping and/or noise is added ontop of the sound signal.

2. This is a really sad result, because timing-wise the removal of the lagcaps leads to finally analogue-like fast response of the filter and amp envelope, as if a blanket was removed from the sound. Clicky clacky on attack and release, and zippy zappy on the filter envelope, exactly as I like it. But just not with all that digital ghost noise and stepping... :-(

3. This very test made me really think about selling my P600 to fund an old analogue synth with hardware envelopes, where non of these problems exist, they always just sound better. Even all the new stuff with digital envelopes, which allegedly are being calculated lightning-fast to sound identically analogue just don´t sound like analogue envelopes to me.

4. Anyone got a dead/cheap P5rev3 for me to restore? Will pay for shipping or gas... ;-)

Anyhow, here are the audio files, enjoy ;-) Headphones may be important...
At the beginning a few seconds WITH the caps, then removal and instant clickyness.
[LAGcaps_CUT.zip](https://github.com/image-et-son/p600fw/files/7900991/LAGcaps_CUT.zip)

Of course I´m contemplating trying to mod the lag circuits in hope that GliGLi refresh rate might not call for such big lags as the Z80, but who knows. And I don´t have smaller caps like 47nf here right now for testing purposes.

(Also I reverted the main board "Rev C-mod" back to original "Rev D spec" and UART 68A50 just to see if there´s a difference timing-wise, but nothing really, no change in glitchyness after power on/seq running. But good to know everything works as before.)

el-folie, 2022-01-20:
Found two 47n caps to put on voice 1, recorded a one-note-sequence, so that every 6th note is voice 1 Listen for the differences. Even 47nf steal the very first precious envelope "click", that analogue is so famous for. Voices 2-6 all exhibit the digital noise. 

Anyway, nice experiments and maybe helpful to some...

[voice1_lagcaps_47nf_voices2-6_nocaps.zip](https://github.com/image-et-son/p600fw/files/7901643/voice1_lagcaps_47nf_voices2-6_nocaps.zip)

matrix12x, 2022-01-20:
The original cutoff for large voltages (thru the 10K resistor when the diodes turn on was 159,155 Hertz)
The 10K resistor with the 47 nF cap gives you a cutoff of 338,628 Hertz.

I wonder what frequency the digital noise is? and why is it on the VCA CV but not on any other CV. For example, the VCF CV and/or the polymod filter envelope to pitch. I'll test mount to see if I get the same noise. I did not notice it in the recordings I posted though.

matrix12x, 2022-01-20:
I'm not hearing the digital noise on mine. VCO 1 no lag caps I did some slower and some faster envelope sounds.

[VCO 1 no lag caps.wav.zip](https://github.com/image-et-son/p600fw/files/7901996/VCO.1.no.lag.caps.wav.zip)

el-folie, 2022-01-20:
The noise is hard to hear with oscillators on and cutoff freq open wide. It gets perceivable using no osc but filter sine and playig around with attack, decay and release of the amp envelope. 

In my examples cutoff is very low and I only used the filter sine to exemplify the digital noise on the amp envelope and make it stand out. No oscillator waveforms were used.

If there´s absolutely no noise on your unit, then it must be the different DAC or other hardware, but it´s hard to imagine...

matrix12x, 2022-01-20:
I'll try it later today the way you describe.
Thats being said, for me, if noise existed only with the filter in self resonant mode and the VCOs off, thats fine, I never use the synth that way.

el-folie, 2022-01-20:
I forgot to say that the noise is also there very prominently in unison sounds WITH oscillators on, it only depends on the attack/release phases then, on filter and amp envelope. In some cases, at the "right" amount of attack/release times there is a weird frequency ontop of the source signal, which I realized is the accumulation of digital steps travelling along the release slopes faster or slower. When the steps travel fast, they become a frequency, when they travel slow they can be heard as individual clicks with noise. 

So it looks like the noise frequency depends on the envelope times and Sequential chose to cut off all digitization noise by filters that are set as low as possible, and at the same time they chose the lag caps big enough at 100nf to never let any digital step becoming audible. Two purposes with one filter solution.

As I only have two 47n caps here right now I can only revert back to 100n. Maybe I´ll put 47n in voice1 only to see if there´s an audible difference to 100n for all other voices. But using the synth without any lag-caps would be impossible for me. Too many sounds I love rely on a clean and beautiful vibrant signal path without digital noise. But for Industrial or Eletro it migt even be benefitial...

matrix12x, 2022-01-20:
Would you mind sending me a single .syx patch that had this noise for you?

el-folie, 2022-01-20:
I will do that syx. tomorrow, had a rough day.

Earlier today I also put back in all the 100n caps. Until we may find a better/faster solution I wanted my p600 being back to factory specs for further OS testing purposes.

I also changed all the 4049 cmos chips that buffer the dac data line signals. I had hoped that one maybe bad buffer chip might be the culprit of the glitches/double triggering, but no, still there. Mystery fault... Maybe it's just timing related, so some clocking circuit on the main board, maybe even just the crystal, timer or divider, or one of little tantals, who knows.

matrix12x, 2022-01-20:
I was able to replicate the noise, but its really faint.
I removed all VCA lag caps. I left in the VCF caps on voices 2-5, the cap was already removed (se above) on voice 1.

I only seem to get some faint noise in unison mode. With and without the VCOs on. It is low enough that I didn't hear it on the synth with headphones in playing at max volume. I had to record it, normalize it and then listen on headphones on my computer. I wonder what route the noise is taking to get in the audio. for example, I wonder if I put extra decoupling caps on the V+ lines on U424 (and the other VCA CV buffer amps). Or I wonder if we left the Cap (for filtering) but shorted out the 100K resistor and lifted the 10K resistor, if that would work.

Of note, I had replaced the 4051 chips in my synth with CD74HCT4051E. A while back I figured the R-ON of the older chips might have gone up with age and I wanted as punchy envelopes as possible.

el-folie, 2022-01-21:
Okay, here´s patch08, 6-voice-unison, typical fat 80s. Please use alpha10 for patch compatibility to ensure correct envelope times for this test.

[patch08.zip](https://github.com/image-et-son/p600fw/files/7909383/patch08.zip)

The noise was on the "note off", so on the release phase of the sound, which is short, but distinctive. The noise was like a high pitched extra pluck on "note off", coming from all 6 voices´ digital stepping noise combined. The noise was clearly audible ontop the source signal, in the P600 state of no lagcaps at all, so 12 caps removed from flt and amp cv lag circuits. It might be harder to hear or not even there with voices 2-6 still having flt caps in place, I never tried that.

Re mains hum and noise floor - yeah, that may mask the digital noise some. My machine luckily is dead silent, noise floor at -100 dB, no mains hum at all.

(My 4051 demultiplexer chips are these: "MC14051BCP". Bought them as new, so I think they might be ok.)

matrix12x, 2022-01-21:
Thanks. I'll check out the noise on the patch soon. 
Yeah I was looking at the recordings you have posted on this thread and the noise floor is much lower than mine, however, the audio files seem to abruptly cutoff at about 18KHz (as if there was a brick wall filter). is that due to sampling rate of your recording device? 

Also, what do you have the VCF limit set to? 

Initially, I was thinking about adding bypassing to the TL082s (i.e., caps on V+ to gnd) that act as buffers out of the D/A (e.g., pin 8 on U424) to clean up any noise.

So, last night I put a cheap oscilloscope on pin 3 of U469 (remember I have C446 lifted) of the lag circuit for voice 1, to look for any noise on the ADSR CV and it looks really clean. I was expecting to see some noise. The release did look a little "steppy" if I really zoomed in. I'll snap an image of the envelope voltage on the scope and update this post later.

matrix12x, 2022-01-21:
After much testing, I realized it's not digital noise. Its the VCA not closing all the way on your unit and slightly less so on my unit. I think theses related to the deadband.
[short release.wav 2.zip](https://github.com/image-et-son/p600fw/files/7916252/short.release.wav.2.zip)

el-folie, 2022-01-21:
Re 18kHz - on my analyzer the P600 looks fine. Most of my sounds don´t have high cutoff though, that might be a reason. VCF limit is off.
[image]

As for the high pitched noise in your audio file on note 3 - that´s not what I have on my P600.
Attached again is the voice1 without flt/amp caps audio file:
[voice1_amp&filter_CV_without_LAGcaps_leads to digitization noise.zip](https://github.com/image-et-son/p600fw/files/7916582/voice1_amp.filter_CV_without_LAGcaps_leads.to.digitization.noise.zip)

Description per time stamp:
0-9 seconds: caps in place then removed
At 10 seconds in the short decay on every note on is accompanied with digital noise ontop of the filter sine
At 14 seconds in the noise blends out shortly (ADSR parameters altered)
At 15 secs the noise comes back
At 17 secs it blends out again
From 18 secs on there is digital stepping noise from dialing in & out sustain level a few times
At 29 secs in the pitched digital noise kicks in badly again
At 30 secs the pitched digital noise shortly blends out
From 31 sec on the high pitched digital noise is on every note on ontop of the filter sine, it sounds almost like distorted aliasing
At 37 secs in the noise is shortly blending out
From 40 to 41 secs very bad digital stepping noise can be heard from moving sustain level again
At 48 secs very prominent noise bursts ontop of the filter snaps
01:16 to 01:25 again very bad digital steps of the filter sweeps audible, it sounds granular, almost like sand/small stones rattling
01:26 to 01:27 on some of the note ons distinct noise bursts can be heard which are not part of the filter sine (these also came in at certain ADSR settings every time)
From 01:29 to 01:39 the digital noise can be heard very prominently and constant ontop of the filter snaps, just a short pause at 01:33 to 01:34.
Exactly on 01:39 is the short release with the "extra noise burst pluck" that I got on patch08, the patch that I sent you. The noise pluck was there on every note off. With the lag-caps it´s gone, together with all the other noises above.
From 01:43 each and every filter snap is layered with "aliasing-like" and granular noise bursts, even the ultra short decay plucks until the end of the file.

Okay, hope this helps to clarify what noise I had without the caps. And this was only one voice in unison with the seq running. At 6-voice-unison all that noise multiplied and it was an instant no-brainer to me to out the caps back in (at this point of time due to further OS testing)

Have fun listening again. :-)

matrix12x, 2022-01-21:
Listened to your audio file. Yeah, thats really different than what I was getting. I tried out patch 8 and was getting something on the note offs. "extra noise burst pluck" is a great way of describing it. But, wow, are those snappy sounding envelopes!

I want to try and figure out the mechanism that is causing it because if we can resolve this, we can (hopefully) get super snappy envelopes.

el-folie, 2022-01-21:
Exactly - if we could somehow get these exact snappy envelopes without all the digital noise (stepping & pitched white noise depending on envelope decay speed) then we´d have an almost perfect synth!

It´s just amazing how extremely fast and snappy the P600 can sound without the lag caps. That´s how analogue envelopes naturally do sound and they don´t need these damned lag circuits.

Yeah, maybe we can find a way to somehow modify or better put, adapt these lag circuits so that they are better tuned for the Teensy refresh rate. Though I don´t know how much faster it is and if it would make an audible difference in the end. But yes, generally the P600GliGLi is very well capable of "the pure snap", it´s just the cv filtering holding it back...

If you as an EE can figure this lag filter riddle out, of course I´d be absolutely delighted. ;-)

matrix12x, 2022-01-21:
I'm trying out some 10NF caps right now. did the 47nF get rid of the noise on the single VCO for you? (I know you didn't have enough for all the voices.)

[image]

I think the detuned saw makes it harder to see the response time. here is a single saw 10nF cap:
[image]

el-folie, 2022-01-21:
47n got rid of the noise but I didn´t do sustain movements to look out for audible stepping. 

Find attached again the audio file with 47n on flt and amp compared to voices 2-6 without caps (and noise). Every 6th note in the sequence running doesn´t exhibit the noise, that´s voice 1. 
But one can also clearly hear that voice 1 completely misses the sharp click of the attack that voices 2-6 do have due to no lag caps at all. 

I guess it´s about finding the balance between noise and snappiness for those lag circuits. Sequence is 6 keys in a row, so voices 1-6.

[voice1_lagcaps_47nf_voices2-6_nocaps.zip](https://github.com/image-et-son/p600fw/files/7917062/voice1_lagcaps_47nf_voices2-6_nocaps.zip)

matrix12x, 2022-01-24:
So the SCI Six-track I have has the following lag circuits:
Looks like the same cutoff freq as the 10 Kohm 0.1UF circuit.

[image]

[image]

matrix12x, 2022-01-24:
Also, in the audio file you posted you can 100% see the difference between the no cap and 0.47uF. I agree, it will definitely be finding the lowest number that still gets rid of the noise yet allows a snappy enough attack through.

el-folie, 2022-01-24:
Wow!!! What a great research you did there, impressive! Definitely the right way to go, finding the right lag timing to mostly still get the snappiness with maybe a tiny amount of noise from stepping as the trade-off of digital envelopes.

And cool you have a Six Track, never had one but was tempted shortly before I got my P600.

el-folie, 2022-01-24:
1. Six Track: 1M and 1n = 159.15Hz cutoff filtering
2. P600 Mod: 100K and 10n = 159.15Hz cutoff filtering
3. Both filters deliver the same time constant of 1ms

Which influence does the difference in resistance have in this case? Maybe none at all?

And, if the above mod with 10n would work for the VCA, does the VCF lag circuit maybe introduce a problem in noise/stepping due to the switch to 10K/10n = 100µs?

Anyway, I guess I need to find 10n caps to listen to. This is exciting!

matrix12x, 2022-01-24:
There is a minor difference between the VCA circuit in the P600 and the six track. The P600 has diodes and a 10K resistor in parallel with the 100Kohm resistor. I modeled this difference.for these quick stack sounds, the 10K ohm resistor is really in play, more so than the 100K was.

Also, the refresh rate on the Six-Track envelopes is the old Z80 style, so it's slow (I didn't measure it), so I'm assuming the digital noise on the Six-Track (if you removed the lag circuit) would be at a different frequency than on a gligli based P600.

el-folie, 2022-01-24:
Ah, I made a mistake above: There is no 10K on the VCF CV, so my question was wrong.

What I also wanted to ask is if the difference of the resistor size (1M vs 100K) does have any impact on the end result, when the filter constant and cutoff are the same?

I guess I just need to listen to the results as that´s what counts in the end...

I found some 10n caps but they are square ceramic types. Bad or should I just try them? Only found two 10n film Siemens caps...

matrix12x, 2022-01-24:
In theory there should be no difference in filtering results. So, in all these cases we are discussing, you could just use a lower resistor if the caps are not available. 

I think they used 1M so that they are not driving as much current... maybe? The p600 was made in 1982 I think, and the Six-Track was made in 1984 or so, so I figure maybe they learned something. I can plot current for these graphs as well. let me know if I should. 

Also, I wonder if smaller cap values are easier to manufacture with lower loss of if higher values are easier (circa 1984).

I think ideally you would want to use film. in throw ceramic caps can have a non-linear performance when charging. Since we are looking for fast transients here, I would go with film.

For the filter lag circuit, I would use a 1nF cap or if you only have 10nF caps, I would parallel up a 10K (or 12k) with the 100K.. or just swap out the 100K for a 10K. I don't believe the exact value would be that crucial.

el-folie, 2022-01-24:
I see - thank you! No need for further graphs, but absolutely feel free if you´d like to explore this topic further.

I´ll have to order new parts then and experiment...

matrix12x, 2022-01-24:
Also, just wanted to note that the filter CV (P600) in has two diodes in parallel with the 100K ohm resistor, and for this fast attack stuff, the diodes effectively bypass the resistor, so the limitation would be the capacitor for this circuit.

el-folie, 2022-01-24:
For the last hours I did some extensive testing and listening until my ears fell off. I installed two precision sockets for the two lag caps of voice 1 and experimented with 100/10/1n caps for VCA and VCF and mixed caps, and compared voice 1 to the other standard voices in running sequences and playing by hand.

My conclusion so far is - if no digitization noise is the goal, which it is also for me, then there is no way around 100n caps for both lag circuits.

Here is why:
Smaller VCF lag caps lead to a better "filter click", which is nice. But this also comes with the consequence that on high frequency filter sweeps the individual digital steps along the decay/release become audible. I tried 10n and 1n and the steps are audible. Only 100n almost completely masks them as tiny variations of the frequency sweeping down. Also, when tweaking VCF sustain level, the smaller the cap the more edgy are the audible digital steps.

Smaller VCA caps lead to a better VCA click, which I wanted badly. But these are only satisfying without any lag cap at all, at least to my ears, 10n already smear the click and also introduces digital stepping noise as a "high pitched pluck" on short ADR times. Especially the R times in unison become too apparent to me as they are already there with 100n caps, just quieter. Also, when tweaking VCA sustain level, anything smaller than 100n leads to very prominent sounding " kr-r-r-r-r-r-r-r-r-r" steps.

So, this is really tricky stuff and apparently Sequential just got it right if the goal was to kill almost all and any digital noise. As for myself I´m not sure how to approach this further as I don´t want any digital noise in my analogue audio signal.

Still, it would be interesting to hear how the Six Track behaves with the lag filter of 1M/1n for VCA and VCF. Maybe it really is a better implementation in suppressing digital noise, as like you mentioned Sequential may have learned something along the way and made the filtering better in every new iteration.

Anyhow, this was/is quite a journey, and an interesting one!

matrix12x, 2022-01-24:
Very interesting. I am always impressed by your testing.

re: the Six-Track, its still the old Z80, so it has the slower refresh rate. It would not be an apples to apples comparison.

I agree there is always some amount of smearing the second you add a cap. (as is seen in the simulations above)

Regarding to the original proposal of adding a software "hold" for a few ms after attack. I think this would give us a little punch. 

Of note, I'm not sure that tweaking the sustain level is a good test, because it's an abrupt level change, and that would inherently have to cause steps. The larger cap would smear that and smooth it out, for the exact reason the large cap kills the attack.  Also, the sustain normally does not change like that in a sound, the ADSR has no way of rapidly changing the sustain. Oh wait, this makes me wonder though, how the random waveform modulating the VCA (which is kind of what you are doing) would sound?
 
I wish we could better quantify the digital noise, then I would have a better plan of attack. For example, is it aliasing noise? is it quantization noise? and what frequency is it? 

I am assuming the digital noise is related to the refresh rate of the ADSR CV.

The original Z80 had an update of about 200Hz (5mS) (according to the service manuals) on the envelopes (for both the P600 and for the Six-Track). The cutoff the Six-Track lag circuit had was about 159Hz. 

I would have figured that since GliGli pushed up the frequency of the refresh rate of the envelope CV (from 200Hz to 2KHz), that we could up the frequency of the lag filter by an order of magnitude. Thus, just making a change to 10nF, in theory should not yield noise.

matrix12x, 2022-01-24:
Interesting to note what they used on the prophet 2000 (one of the last synths they made from that time period):
VCA looks like a 24K and a 0.1uF cap with a 330ohm and a diode in parallel with the 24K.

More interesting is the circuit on the VCF CV input.

[image]

el-folie, 2022-01-25:
As my P600 is now prepared with sockets for the lag caps on voice 1 I could do any kind of audio example of noise you might need for research on noise frequency and stepping. If you just need random audio files in which the noise becomes most apparent I can also do those, with and without caps in place.

Re. Prophet 2000:
VCA fast switching speed CV = (390Ohm/100n) = 0.039µs (which presumably is like no cap at all)
fc = 4.08kHz (noise will propbably ocurr)

VCA normal CV speed = 2.4ms (which is quite fast already)
fc = 66.31 Hz (which probably damps any noise)

The VCF lag circuit looks quite a bit more sophisticated. I would interpret it like this: 
20K/2.2u is the standard low pass filter, but extremely slow at 52ms (fc = 3Hz, no noise). 
This is made faster by the 1K in parallel to 2.2u, because it presents a path to ground and changes the frequency cutoff of the filter (I know this from tube amps where a cathode resistor in parallel to a cap forms a filter to influence the tube´s response). The 1K also forms a voltage divider with the 20K. And then there´s also the -5v via the 33.2K at the junction of the general low pass filter. For me it´s very hard to say what might happen there in real-time... ;-) Do you have any clue?

el-folie, 2022-01-25:
> Regarding to the original proposal of adding a software "hold" for a few ms after attack. I think this would give us a little punch.

Yeah, we got a bit distracted with lag caps and things... ;-) Of course the Attack Hold parameter is a different thing completely and would definitely add that Moogy punch, which comes after the envelopes clicking anyway.

image-et-son, 2022-02-03:
Hi, now I see what you meant when you said you got "a little distracted". I completely missed that discussion until  I just now saw that there 54 posts on the ticket. What's the bottom line? 

I caught some posts on VCA not closing completely. As I changed the deadband value slightly to match my instrument nicely (it really fades out into nothing on my P600 now) I'd be interested if other instrument need another value.