image-et-son/p600fw #66: Collective bug for UI navigation errors / improvable points

image-et-son, 2022-01-10:
1. When in record mode pressing arp buttons should cancel recording mode (like for sequencer). Currently the function sequence is odd: press record (record mode, record blinks), press arp U/D (arp U/D blinks, but overall still in record mode), press record (record goes solid, in arp hold mode), press arp U/D (arp U/D goes out, record start blinking, record mode). Open question: does the sold record LED in seq edit mode always also indicate hold mode for arp? The meaning of the record LED should be unambiguous.
2. Arp should start directly into hold mode

matrix12x, 2022-01-10:
I think I came across a UI bug just now, I think this is the correct spot for it, if not, let me know and I will split this off:

Parameters: in preset mode, playing any arp with any arp button pressed (up/down or random, etc). When holding the sustain pedal (arp latch) Record Led turns on (like it should), I then press the same arp button to turn arp off (but I'm still holding the sustain pedal), and here is where the issue happens: record light stays lit, no matter what, because I did not release the sustain pedal before pressing the arp button.

ex: 
Press ARP up/down to start arp (Arp up/down led On)
Press and hold Sustain pedal (Record led turns On)
Press ARP up/down (to turn off arp)
Record led stays lit no matter what I press next. Can't turn it off without switching on and off P600.

Not sure if this bug matters, I mean who besides me keeps sustain pedal held when turning off the arp. It was a fluke.

RDPRDP, 2022-01-12:
When I try to use pickup mode many times I cannot pickup the knob, maybe the conditions are too tight? Also, I think this happens more mid sweep of a knob more than when the pickup point is at the start or end.

image-et-son, 2022-01-12:
@RDPRDP: thanks for posting this. To nail it down I have a few questions: 1) you had this behaviour in different occasions on different dials? 2) Even very slow approach did not "pick-up" the dial? 3) Did this affect bipolar (+/-) dials specifically?

image-et-son, 2022-01-12:
> I think I came across a UI bug just now, I think this is the correct spot for it, if not, let me know and I will split this off:
>
> Parameters: in preset mode, playing any arp with any arp button [...]

Fixed it. I'll mark the individual point in the issue text as "DONE". I don't want to have too many small ticket flying around. Thanks for posting. It was a bug.

matrix12x, 2022-01-12:
> When I try to use pickup mode many times I cannot pickup the knob, maybe the conditions are too tight? Also, I think this happens more mid sweep of a knob more than when the pickup point is at the start or end.

I am having the same issues. I just thought it was me and that even slowly, I may tweak too fast. Generally, I think its too tight for the speed at which I tweak the knobs (my slow). I can eventually get it, but it takes me 4 or 5 passes left and right on average.  I have this on all knobs, not just bipolar.

RDPRDP, 2022-01-12:
> 

1. Different occaisions on different dials.
2. When I have the issue, even slow doesn't help.
3. Any knob that has pickup can have this, doesn't need to be bipolar.

I see Matrix12x has similar issue. I also sort of "sweep" the knob trying to get it to pickup, and this often works after a few times of back and forth, at slowly reducing speeds, but sometimes I cannot get a know to pickup, not matter how gentle I am with it. Of course reproducing this issue on demand is quite a challenge....  ;-)

matrix12x, 2022-01-14:
Alpha 11
I confirm my Arp/sustain pedal bus is fixed. 
Knob pickup is way better. I like it.

image-et-son, 2022-01-14:
I'll create a new collection for upcoming alphas

closed 2022-01-14 by image-et-son