image-et-son/p600fw #69: Feature request: Spread and VCF Limit under CC or SYSEX control

RDPRDP, 2022-01-12:
If not already implemented it would be nice to have Spread (Full CC range, or SYSWEX) and VCF Limit (CC 0-63 OFF, 64-127 ON, or SYSEX) under midi control.

matrix12x, 2022-04-06:
I was thinking about this and I think this is best handled as sysex. It's not a thing one would change in real time, and it is bets to preserve CCs for real time control.

image-et-son, 2022-04-07:
> I was thinking about this and I think this is best handled as sysex. It's not a thing one would change in real time, and it is bets to preserve CCs for real time control.

Yes, we're running out of CCs anyway. Maybe it would be possible to implement a general SysEx format for settings.