image-et-son/p600fw #70: Take tuning into account when applying poly-mod pitch shift of VCO A

image-et-son, 2022-01-12:
At the moment the frequency CV equivalent from poly mod is simply added to the tuned oscillator CV (see function refreshVoice() around lines 940 etc.). The shift can be large, though. Depending on the difference in chip characteristics, in particular, variations across the frequency/voltage range which are compensated by tuning, this can lead to strongly untuned shifts. The change will affect old patches, but it is fair to assumed that almost no user has actively worked with the imperfections of this, in particular since this is also voice dependent.

image-et-son, 2022-07-15:
Horray, a new synth is born. Completely new patches are possible due to harmonic shifting. For those eager to try out: try the two patches with an older version (sounds horrible) and then with the attached release candidate 1 (sounds harmonic).
[p600firmware_v2022_RC1_20220715.zip](https://github.com/image-et-son/p600fw/files/9122173/p600firmware_v2022_RC1_20220715.zip)
[2_patches_with_harmonic_shift.zip](https://github.com/image-et-son/p600fw/files/9122179/2_patches_with_harmonic_shift.zip)

matrix12x, 2022-07-15:
I tested both patches on the prior Beta and they don't sound too good. I updated to the version you have posted above and they sound in tune. Patch 01 sounds amazing.

closed 2022-07-16 by image-et-son