image-et-son/p600fw #71: Still glitchy behaviour of LFO Amount, PWM A & B and Poly-Mod Envelope (in combination with envelope sustain)

image-et-son, 2022-01-12:
There is still strong steppiness, in particular  in the PWM A & B dials and in the LFO amount. Those dials which have a strong impact on pitch (frequency A & B, for example, but also other dials depending on patch) have reached a maximum time resolution now. They are already in priority mode. For the non-priority dials, there should be a dynamic focus mode so that fast movements are as smooth as possible.

matrix12x, 2022-01-13:
I'm getting a weird reduced response in PWM A. PWM B seems to be able to "go through zero" on both ends of the range (min and max), but PWM A seems to have big dead spots. However, they both sound smooth and I'm not getting any steppyness. Compiled 1520cc8. 

LFO Amount seems smooth.

image-et-son, 2022-01-13:
I tried it, and for me PWM A and PWM B work identically...!? There is also nothing in the code that would treat the two differently.

matrix12x, 2022-01-14:
Checked it again today, so it works fine in preset mode, but strangely in manual mode the PWM A has a weird reduced response range from about 00 to about 40 then the actual pulse width goes through zero and stays there.

The LFO to PWM switch is off. LFO amount is zero. I'm not sure what other parameter can screw with the range on PWM A  only but not PWM B. and I agree, I don't see anything in the code. I wonder if I should re-flash using the .hex. I've done the last 5 or so flashes via .syx. I wonder if there is "garbage" in the memory.

matrix12x, 2022-01-14:
Good point re verifying prior version behavior. I will verify behavior vs the alpha 10 later today.

"Fine in preset mode" means PWM A and PWM B behave as they always did. The displayed values are still 00 to 99 even in manual mode. What I mean is the actual amount it is changing the pulse width for VCO A in manual mode is seemingly reduced. like above 40 there are no audible changes and at a PWM A of 00 I'm not at "silence" For VCO A's pulse as I was in preset mode.

If nobody else has this issue, it may be some form of user error (me).

matrix12x, 2022-01-14:
Synopsis: no actual bug.

I tried it out with the alpha 10 and was getting the same issue so I refreshed alpha 11 using the hex file and .... 
Good news, no longer do I have an issue.  

The issue was really weird. it was almost like sync was flipped on, but it was not.

So that being said, there is no actual issue. it was my synth being weird.

closed 2022-01-24 by image-et-son