image-et-son/p600fw #72: Speed dial to clock/speed on start-up and patch selection by dial (in preset panel mode)

image-et-son, 2022-01-13:


matrix12x, 2022-01-14:
I confirm that on Alpha 11 this works.
on boot in manual mode speed knob is speed. I press "preset" and speed knob defaults to patch selection. 
of note if I hit "to tape" then go to manual mode, it stays as if  "to tape" was still lit. not sure if this is intended

matrix12x, 2022-01-14:
I agree that changing presets via midi and the knob should not write to flash (I.e., should not save the selection).
I just think we should clear the "dot" in both cases.

image-et-son, 2022-01-14:
> I confirm that on Alpha 11 this works. on boot in manual mode speed knob is speed. I press "preset" and speed knob defaults to patch selection. of note if I hit "to tape" then go to manual mode, it stays as if "to tape" was still lit. not sure if this is intended 

I could not reproduce this. But maybe I didn't the exact thing that you did. Can you please clarify? (the code is such that things like this can easily slip in, sometimes with unpredictable outcome, sometime miraculously doing nothing bad...)

matrix12x, 2022-01-14:
Good news, I can't get it to replicate today. at all, under any circumstance.
Who knows what actually happened.

closed 2022-01-14 by image-et-son