image-et-son/p600fw #73: Remove LFO range parameter (and rescale old patches)

image-et-son, 2022-01-13:


el-folie, 2022-01-13:
Just an idea: 
Maybe the same method would be applicable to the #3 mod wheel range, where you fine-tuned the modulation onset. By reducing all those different options to one nice smooth onset and going up to max in the last third of wheel travel we could get rid of all the options to further simplify and declutter the menue for the user. (And make space for other options).

Good idea/bad idea?

matrix12x, 2022-01-13:
I agree with @el-folie

el-folie, 2022-01-14:
The slowest Speed on 0 is still too fast for my taste (like with a slow, nice piercing PWM by LFO). It could be like 4-5 times slower at 0 ideally.

And having the full range on the LFO freq knob is so nice for live tweaking, much much better than before!

el-folie, 2022-01-14:
Yes, absolutely. Maybe I didn´t test that aspect earlier but yes, the onset of Speed should be really really slow to create filter resonating ambiences or slow piercing PWM sounds that travel along the brim of PWM breakup to zero.

image-et-son, 2022-01-14:
Additional point continued in #78

closed 2022-01-14 by image-et-son