image-et-son/p600fw #74: Make the panel layout switchable between Mix/Glide and Vol A/Vol B

image-et-son, 2022-01-13:
Idea: a new misc setting provides the choice between a) the original dial layout with Mix and Glides and b) the GliGli layout with Vol A and Vol B. This would involve differences in the additional patch parameter menue too (see #27).

image-et-son, 2022-01-13:
Functional aspect, proposal (all parameters 0...1):
VolA = VolM * Mix
VolB = VolM * (1-Mix)
VolM = VolA + VolB
Mix =  (VolA - VolB) / 2VolM

And the proposed "Drive" parameter might be: VolM = HALF_RANGE * (1 + Drive). The problem is the following: while it would be very good to have a constant output for every position of "Mix" this cannot be achieved for all values of drive because a constant output of a an equal mix of A and B with both at maximum FULL_RANGE would need to become >FULL_RANGE for VolA as the Mix is turned towards A. Otherwise the output is not constant. One possibility would be a more complex function, where drive up to 0,5 brings either oscillator to FULL_RANGE (as in the function above) but beyond 0,5 the volume continues to stay constant as a function of Mix until at full drive the function approaches: VolA = FULL_RANGE for Mix <= 0,5 and zero otherwise and VolB = FULL_RANGE for Mix >=0,5 and zero otherwise. 

But the behaviour will not necessarily be intuitive.

matrix12x, 2022-01-13:
I'm not 100% sure conceptually how the mix knob would work. If it worked like the old mix knob, you effectively lose the overdrive.

To cure this possible issue, are you proposing replacing the two knobs with 1 knob and 3 menu options?

One of the main benefits of the separate Vol A and Vol B knobs is the overdrive. I think when people comment that the P600 sounds better in some ways, it is specifically because of the overdrive feature. I think you hit on all of the issues in the post above.

I still am a fan of just Vol A and Vol B with glide in the menu. Even on the original Sequential design I think a mix knob was a poor choice. Every patch one makes uses Vol A and Vol B. Only some patches use glide.

image-et-son, 2022-01-13:
I have serious doubts if the two concepts can be reconciled. My formulae above describe the transformation between two parameters in the orig layout (Mix + a new drive menue parameter) and the two parameters in the GliGli layout (VolA and VolB). I don't think you can cover the entire overdriven parameter ranges with Mix + drive, at least not if mix keeps the volume constant.

el-folie, 2022-01-13:
I think in original panel mode it would be great really, if it just behaved like with the Z80, where full A or B are louder than their 50:50 mix in the middle of the mix knob. In my opinion it sounds beautiful and musical on Z80, not really a need for  additional overdrive at 50:50 mix position. It sounds fat and balanced already.

matrix12x, 2022-01-25:
"From Tape" +6 for switching?

image-et-son, 2022-01-25:
Yes, there used to be MIDI dump, I think. Now it has a new function :-)

el-folie, 2022-01-25:
So great! Looking forward to this, :-)

matrix12x, 2022-01-25:
I think I found a bug, in SCI mode "From Tape" + Preset has weird results. it still works fine under GliGli mode

image-et-son, 2022-01-25:
There is some intricacy of transforming internal variables when changing modes. So, there's bound to be something I overlooked or where I made a mistake. I did try it out thoroughly of course. Can you be more specific of what is going wrong?

matrix12x, 2022-01-25:
Mix is set to full volume VCO B and zero volume VCO A.

el-folie, 2022-01-26:
First I need to say this: I got lost for 3 hours straight in just playing the machine due to the new original panel mode. I´ve still got a grin on my face typing this. It´s just so great to havethe orginal SCI panel back together with the new envelopes, fantastic vintage parameter and fine-tune. I got totally lost in tweaking, listening and playing, like a meditation. So first, a big thank you, imogen, for making this possible! :)))

Bug: 
For all presets volume of oscillator B is zero. A little movement of MIX brings it into play.

image-et-son, 2022-01-27:
> Bug: For all presets volume of oscillator B is zero. A little movement of MIX brings osc B back into play.

Found the coding error (really bad mistake on my behalf), thanks. See issue #95

el-folie, 2022-01-27:
Just in case anyone is interested in how the Z80 P600 differs in osc volume compared to alpha13, and to give a hint on how to set the new drive parameter for original Z80 volumes in alpha13 SCI panel mode, I did some tests yesterday night, using the RME digicheck peak level meter: 

P600 Z80:
VCO A max MIX level = -13.3 dBfs
VCO A mid MIX level = -19.0 dBfs
VCO B max MIX level = -12.6 dBfs
VCO B mid MIX level = -19.4 dBfs
VCO AB detuned range from -13.3 to -19.4 dBfs
(And yes, OSC B is louder on my P600 for all waveforms)

P600 alpha13 - Drive at max - GliGli panel mode: 
VCO A max level = -12.5 dBfs
VCO A mid level = -18.5 dBfs
VCO B max level = -11.8 dBfs
VCO B mid level = - 17.7 dBfs
VCO AB detuned range from -6.0 to -12.2 dBfs
(So, GliGli allows for a lot more osc volume and saturation on VCO AB detuned sounds.)

If the goal is to achieve an original SCI/Z80-OS-like saturation response on alpha13 in the new SCI panel mode with MIX AB knob, the new Drive parameter on my machine had to be set at a value of about 44. This will slightly vary from machine to machine of course.

And it´s so nice that by calibrating the Drive parameter we´re now able to use the old P600 sound engine at the exact same saturation levels as originally intended by Sequential and at the same time can go beyond that if we like to. Thank you, imogen! :)

el-folie, 2022-01-27:
As the panel switch mode is being retained in memory now after a power cycle and MIX/Glide are working, this issue could be considered done.

Again thanks - so great to have this feature!

closed 2022-01-27 by image-et-son