image-et-son/p600fw #79: Collective UI Issue for alpha 12

image-et-son, 2022-01-15:
1) When in preset panel mode one twecks a knob which is not licked up, this knob is active. When switching the preset patch mode the knob is then directly applied which may lead to a sudden, unwanted change in this parameter

2) When selecting the default patch, the P600 should definitely switch to preset (panel) mode. Otherwise the default patch is not what the user hears.

3)  ...

matrix12x, 2022-01-16:
Alpha 11.1 (and Alpha 10, I didn't go back and check earlier)
In Manual mode "to Tape" led never lights when "to Tape" is pressed. However, "to tape" is functioning like as if it were lit.
Is this a bug or intended?

matrix12x, 2022-01-16:
BUG as posted in gearspace:

Jpro600K posted:
"program number set [to] 16
to tape + 77 glide 
Set value 76
Push To Tape button
Program has been changed to 76."

I confirm this bug (Alpha 11.1). 
It is not just glide.
If any "To Tape" parameter that has a numerical value is changed when in Preset mode, when "To Tape" is pressed again the preset number changes to the value of the parameter.

image-et-son, 2022-01-16:
OK, I see. Even when fixing that (which is probably easy) I think a pick-up logic is required for the data dial for selecting patches, otherwise you'll regularly just load the patch where the data dial happens to be after you change to preset mode and touch the data dial.

image-et-son, 2022-01-16:
Opened issue #80 for the pick-me protection of the current patch selection

Colpocorto, 2022-01-16:
If a parameter is changed by using the classic UI, the dot is turned on, as expected. But if now you change the program with the data knob, the dot remains ON despite the newly selected program is still unaltered.

el-folie, 2022-01-19:
Alpha12:
When switching from panel to preset mode the last used parameter value is still selectable by the Speed knob, even though To Tape hasn´t been pressed yet to select a parameter from the menue.

matrix12x, 2022-01-19:
Alpha 12
I confirm I have the same issue as @el-folie

Colpocorto, 2022-01-19:
I confirm #89 too (both Alpha 11 and 12).

image-et-son, 2022-01-19:
Thanks, other point in the issue OK?

image-et-son, 2022-01-20:
Hey 

> Alpha12: When switching from panel to preset mode the last used parameter value is still selectable by the Speed knob, even though To Tape hasn´t been pressed yet to select a parameter from the menue.

Hi, I branched off the issue #89 from this. However, when I thought about it, I think this is the desired behaviour. Please refer to #89 for this.

image-et-son, 2022-01-20:
Number 1: TESTED, DONE

closed 2022-01-20 by image-et-son