image-et-son/p600fw #82: Notes are repeated in the up-down arp mode under some circumstances

Colpocorto, 2022-01-16:
If I press a two-hand chord and start moving one or more notes (on the left hand), some of the middle or higher notes (right hand) are repeated (two in a row) without running up and down the whole pattern (the result is sorta random pattern instead of an up & down one).
[newarp.wav.zip](https://github.com/image-et-son/p600fw/files/7877578/newarp.wav.zip)

image-et-son, 2022-01-16:
It appear that notes are being play while the array of notes is being reshuffled, so that notes are skipped or played several times because the index of the notes changes.

Colpocorto, 2022-01-16:
> It appear that notes are being play while the array of notes is being reshuffled, so that notes are skipped or played several times because the index of the notes changes.

I think that Gligli's first approach of using the array index for both selecting the pressed note and keeping its correct order was a strategy to prevent exactly this problem.

image-et-son, 2022-01-16:
Yeah, my idea was to 1) avoid computation time and 2) to avoid complex logic to get the better pattern. But thinking about it now 1) the computation is not an issue any more and 2) I can readily think of a simple way to get the new pattern with GliGli's logic. So I will go back to the array. But maybe in alpha 13 because there are some bugs which need attention now.

image-et-son, 2022-01-23:
Hi, is it ok now in alpha 12.1?

el-folie, 2022-01-23:
I tested it and didn´t have a single case of double triggering the same note in up/dwn arp.

But maybe Mastropiero´s use case was very specific, so best he´d test it. His audio file above sounds totally beautiful and captivating despite or maybe even because of those double notes though. ;-)

closed 2022-01-27 by image-et-son