image-et-son/p600fw #83: Reduce the number of manual page stores

image-et-son, 2022-01-16:
Currently manual parameters are stored every time a new menu parameter is selected. This could be reduced to after a value CHANGE of a different menu parameter. E.g. when a parameter is changed the OS has a flag that says: store it when ever another parameter is changed. In this way the number of storage operations could be strongly reduced.

image-et-son, 2022-01-18:
Now it works as follows: the store is done not when another menu parameter is selected but only the first time this parameter is actually changed. This should reduce the number of stores by about 3 with minimal impact on user convenience. This is more important now with many 3rd press parameters.

matrix12x, 2022-01-23:
What is the best way for me to test this?
Should I just change a parameter in manual mode such as vibrato speed, then change vibrato target switch to preset and back and see if speed and target stored?

image-et-son, 2022-01-23:
The behaviour has changed slightly for the user: before the live patch was stored:
- (before switching to preset mode or before storing the patch or) when a menu parameter SELECTION was changed (that is potentially a lot of storage, and even more so with third press parameters!)

With the new version it is stored:
-(before switching to preset mode or before storing the patch or) when a DIFFERENT menu parameter was changed (that means the users selects a different parameter AND also changes its value)

The consequence for a user is that the last change menu parameter is always lost WHEN SWITCHING OFF. When you change to preset mode it is still stored.

Maybe you could check and assess the situation? What I can do is to monitor the storage execution to confirm the new storage logic from that perspective.

image-et-son, 2022-01-23:
From the storage perspective it works as designed. I guess we can close this and leave the assessment of whether this is a good /acceptable idea to the beta testing phase.

closed 2022-01-23 by image-et-son

matrix12x, 2022-01-24:
I confirm it works as explained above.