image-et-son/p600fw #86: Proposal: Exact middle position for PW knobs on osc A and B for PW @ 50 %

el-folie, 2022-01-18:
On my P600 with exact 4.90v DAC gain it´s like this, maybe your P600s behave a bit differently:

50 % pulse width is not in the PW knob middle position at value 49 (= middle between 0 and 99), but exactly at value 47 (by ear). Because of that the PW cutoff values of 0 % and 100 % are unsymmetrically distributed on knob position values 0 and 96, which makes the PW already appear between value 0 and 1 and after PW 100 % there are 3.5-4 knob values of silence from 96.5 to 99.

Another PWM thing:
Maximum LFO PWM modulation by initial amount doesn´t reach the PW cutoff points. The mod wheel goes a tiny bit deeper in mod depth but only reaches the cutoff point at 100 %, not at 0 %. 

Symmetrical PWM cutoff modulation is a nice effect to use and needed for some sounds. So it would be great to have a bit more modulation depth for PWM on initial amount and the mod wheel depth. Also, a congruent middle position of the osc A and B PW knobs at 50 % pulse width would be very nice, even if just for visual clarity.

el-folie, 2022-01-18:
Adding a nice musical audio example with and without PWM cutoff for comparison and clarification. (I was on Z80 for this) 

[Desktop.zip](https://github.com/image-et-son/p600fw/files/7889327/Desktop.zip)

image-et-son, 2022-01-18:
OK, the general idea would be to
- first: re-scale the range from [0...95(?)] to [0...99] but in order to understand; and
- second: increase the range of mod wheel and initial amount

In my case the PWM breaks up at around 92 (on A and B). So I ask myself: where do I aim, 96.5 as in your case or 92 as in my case? What variation is there among the P600 out there?

el-folie, 2022-01-18:
To 1st:
Rather the opposite - rescaling so, that around 1 and 98 already is silence to make sure cutoff can be deliberately dialled in.

Maybe it would be best as a first step to measure the DAC gain on your P600. Maybe it is off the 4.90v mark and so the difference of 92 to 96.5. But if it is at 4.90v too, then every P600 will be different. In that case, if possible, we´d need a user setting to calibrate the PW knobs at 50 % middle position, like with the PB wheel (or combine it with the PB wheel calibration?). Or we forget about these little differences and see them as "it´s analogue - so what".

To 2nd:
Yes, just a small increase at the top values of MW and Initial Amount to reach PWM cutoff.

image-et-son, 2022-01-18:
Yes, I think I meantthe same thing about the rescaling and I also thought about a calibration procedure. For that we'd need a definite description of what the correct behaviour of PWM is. Can you tel

el-folie, 2022-01-18:
Give me a sec...

el-folie, 2022-01-18:
From the Technical Manual:
[image]

el-folie, 2022-01-18:
You don´t need to power off to measure though, also it´s best to measure when the unit is warmed up and all voltages have settled. U246-7 is the OSC A CV. So on GliGLi just turn OSC A volume full up and then calibrate for 4.90v at pin7. Any ground is fine, I just use a screw of the analog board.

image-et-son, 2022-01-18:
I have 4.99 

Should I trim this to 4.9?

el-folie, 2022-01-18:
Interesting! Maybe the voltage moved up in 40 years...

Yeah, I´d definitely calibrate it to 4.90v per Sequential specs and then see what happens to PW knob middle position and range, would be interesting if it then settles for the same values my machine spits out.

image-et-son, 2022-01-18:
A small change, not much. The break may moved up to about 93 on my P600. Pretty consistent on all voices. 

What I also found is that my exact tune point between A and B on the fine adjustment is at about +3.

image-et-son, 2022-01-18:
I wonder if it makes sense to extend the bender calibration to fine tune, PWM A Start, PWM B Start, PWM A End, PWM B End, VCA Deadband...

el-folie, 2022-01-18:
Maybe after DAC calibration also the scaling process is necessary? I´d imagine that scaling might have an influence on the fine-tune knob for example (and PWM).

el-folie, 2022-01-18:
> I wonder if it makes sense to extend the bender calibration to fine tune, PWM A Start, PWM B Start, PWM A End, PWM B End, VCA Deadband...

And absolutely, if at all possible it would be great to fine-tune each individual machine to exact specs.

image-et-son, 2022-01-18:
Is that goo there from factory? I have it too and I assumed that someone thought he did a "final" job. My feeling tells me that I should rather concentrate only finalising the code ...

el-folie, 2022-01-18:
Yes, it´s factory goo ;-)

And yes...

matrix12x, 2022-01-18:
For VCO A the PWM zero point on my P600 is at 00(low) and a hair over 94 (high). My PWM 50% point as measured using an oscilloscope is about 47.

For VCO B my midpoint is measured at 49, my zero low is 00 and high zero is low 93.

I think the software PWM mid point based on all three of our synths can be nudged up a few to 50.

Also as noted at U467 Pin 7 when VCOA's volume pot is maxed out (99) I measure 5V. At about a pot value of 96 I get 4.9V. I have BB DAC71 and cannot adjust the DAC gain. I am sure there are others with the same DAC 71 that do not have the ability to adjust DAC gain. This seems to match the results @image-et-son measures on his (but he has a trimer). I still think that the max voltage out at U467 pin 7 on the Z80 code was 4.9V, i.e. the code was not going to max.

My P600's VCO A and B fine tune midpoint as measured on an Oscilloscope is 00 (slightly above 00, but still displaying 00)

el-folie, 2022-01-19:
@matrix12x

You are correct! When calibrating DAC gain on Z80 as per Sequential spec to exactly 4.90v, the voltage resulting with GliGLi (without touching the trimmer) is 4.96v to 4.97v (fluctuating on my simple DMM). So GliGli really upped the filter input saturation like you assumed.

That also means that DAC gain calibration with GliGli must be set 0.6 to 0.7v higher than on Z80, if we assume that the original 4.90v at U426-7 were meant to be exact to keep sure that other voltages will then be in a specified range. All in all it really shows that DAC gain is not critical, as being said so in the Technical Manual. 

Ok, need to sleep now, almost 2 am here.

image-et-son, 2022-01-19:
In my case the difference between 4.99 and 4.9 was minimal, if any.

BTW: I don't know what the state of my P600 was, probably human error, but the exact fine tune is close to zero position. So please ignore my info from before. This means that voltage at U426-7 really has almost no impact (I guess within a certain range :-))

matrix12x, 2022-01-19:
Alpha 12:
 On my P600 my VCOA PW and VCOB PW nudged up nicely. The low end cutoff on my unit is pretty much 01 for both and the high end is pretty much 97. My mid point is about 50 for both.

Very nice!

el-folie, 2022-01-19:
Alpha12:
On my P600 nudged up by two values to midpoint at 49, much nicer than before! 50 would be even nicer as on my P600 low cutoff is at 0.9 or so, hi cutoff at 98. So, one notch up would be perfect here.

(Still can´t fully cutoff hi/low PW by LFO modulation MW or Initial knob , but I guess this wasn´t a part of alpha12.)

image-et-son, 2022-01-19:
Extending the (PW) range of LFO wasn't in alpha 12. But, I am worried about the variances in P600s out there. Currently I have another 3 people I think I can trust to take this seriously  to look into it during a beta testing phase (ideally I'd like to have something like 10 people in beta testing). I would like this software version to  work on all (normal) P600s. Unfortunately it doesn't work on yours in that respect., it seems

In my case I can easily reach break the up point on both sides  with LFO plus some head room. In the code the full range of LFO is added to the PW, so I would not really know how to increase it, because it's already maxed out. You are saying that you could reach either side (by using also the PW A&B pots) but not both?

el-folie, 2022-01-19:
I can reach full PW cutoff by the PW A and B knobs on either side.
I can´t reach full PW cutoff on both sides (0 and 100 %) by LFO modulation.
I can reach either 0 or 100 % cutoff point by LFO modulation, but not both at the same time with the PW A knob at exactly 50 % duty cycle.

el-folie, 2022-01-19:
Maybe an issue of DAC gain? Maybe I should try the 4.99v you had earlier...

image-et-son, 2022-01-19:
It could be that for some reason you can't max the LFO amount by pot? Could you please use the single patch dump and send me the syx? You would ne to stored it in alpha 12 first for that.

BTW: I have written a python script which displays the parameters of a syx patch. It is here: https://github.com/image-et-son/p600fw/tree/master/syxmgmt ( the *.px script requires the 2 storage specs,)

But you can also send it to me.

el-folie, 2022-01-20:
Sorry, got lost in testing...

1. The mod wheel on my P600 definitely goes a tiny bit deeper in modulation depth than the Initial amount knob can. 
2. I was finally able to fine tune PW A B so that with the mod wheel both cutoff points were reachable by the LFO.
3. (The sound in my head is a bit more choppy still, but this is okay)

How do I do a single patch dump? Never tried that before... ;-)))

image-et-son, 2022-01-20:
How do I do a single patch dump?

Connect MIDIOX. Press FromTape + Record. The Preset Button blinks. Then type the number of the patch and it will exported as sysex and you can see it in MIDIOX on the input monitor!

el-folie, 2022-01-20:
Saved a simple PW cutoff sound with Initial amount on 99.
Mod wheel then goes to 11, so to speak ;-)

[PW cutoff by LFO Initial amount 99.zip](https://github.com/image-et-son/p600fw/files/7901366/PW.cutoff.by.LFO.Initial.amount.99.zip)

image-et-son, 2022-01-20:
Ok, I will look at it tomorrow - it's late in CET

el-folie, 2022-01-20:
CET here too - Good night 👍

image-et-son, 2022-01-20:
I checked your patch and the LFO Amount pot value really is 65535 (max value) and the two PW pots pretty much center. The maximum amplitude of both, LFO and vib, is currently half a pot range (so about 65535/2) which at max LFO Amount and centered PE pots should theoretically take you from 0 PW to 99 PW (internally: 65535). I could extend the LFO range, of course. I think for pitch and filter cutoff the range is (more than) enough. For PWM you could argue that from zero to zero is also enough. But I think it is important to define a behaviour "according to spec" and it is unsatisfying that you can't reach that. It is even more surprising given that the typical PW pot range is more than from zero to zero.

image-et-son, 2022-01-20:
> Saved a simple PW cutoff sound with Initial amount on 99. Mod wheel then goes to 11, so to speak ;-)
> 
> [PW cutoff by LFO Initial amount 99.zip](https://github.com/image-et-son/p600fw/files/7901366/PW.cutoff.by.LFO.Initial.amount.99.zip)

The single dump function is much more convenient than in v2.*. Please find enclosed the values of your patch!

I have written python script "syx_converter.py" to produce it. To use that, download the *py files and the two *.spec files. When you copy a syx file in to the same folder you can on a command prompt execute ">python syx_converter.py [syxfilename]" and it will display the values as in the file above. You also have and option "-p[patchnumber]" if you only want one particular patch from a bulk syx.

[patch75.txt](https://github.com/image-et-son/p600fw/files/7904607/patch75.txt)

el-folie, 2022-01-20:
So, extending the max LFO range just for PW modulation would exactly do the trick then. The sound I tried to achieve is a symmetrical lo and hi cutoff of PW by LFO, where the cutoff duration is clearly audible as a little pause. So I guess if max PW modulation would already be reached at LFO Inital amount of 95 (or so) the last few values could be used to dial in an exact hi/low cutoff period.

Thank you for the patch txt! Will also wrap my head around the python script, right now I´m "un-gooing" the P600 trimmers for later scaling and better testing.

image-et-son, 2022-01-24:
While working on the OSC B fine frequency pot (#85) I became aware that I was doing the wrong thing with the PW pots recalibration: I programmed a function which scaled the patch PW value to the one the should be applied as a voltage. Instead I should be rescaling the pot value. In that way old patches also don't need to be rescaled for PW A & B. I will need to reprogram this, so back to work in progress. (Note: it will be a generalized deadband/guard function.)

image-et-son, 2022-01-25:
This needs some more thought because it is a hardware topic, e.g. the PW voltage range is neither centered not does it cover the entire range. What is needed is a survey of the variation of hardware. In my case I find

  | Start | Mid | End
-- | -- | -- | --
A | 505 | 29612 | 60394
B | 381 | 29301 | 60083

el-folie, 2022-01-25:
I see, I was just going to ask how to read out those values.

Does start/end mean "beginning/end of audible PW" or "increment before/after audible PW"?

el-folie, 2022-01-25:
Installed python 2.7 to read out syx patch files. I´m on Win10, what am I doing wrong?:
[image]

image-et-son, 2022-01-25:
> Does start/end mean "beginning/end of audible PW" or "increment before/after audible PW"?

Yes

el-folie, 2022-01-25:
I was incapable to get the python readout script working, so here are my PW Start-Mid-End patch dumps. There´s variation from voice to voice, so I chose the voice which sounded first/last on PW knob travel.

[PW_AB_START-MID-END_P600_el-folie.zip](https://github.com/image-et-son/p600fw/files/7935446/PW_AB_START-MID-END_P600_el-folie.zip)

image-et-son, 2022-01-25:
Thanks, here's the result of your P600 (assumed that you did this on alpha 12.1):

  | Start | Mid | End
-- | -- | -- | --
A | 198 | 31228 | 61824
B | 260 | 31353 | 62010

Quite different. :-\  - not only the mid position but also the range. Some patches will sound different on different machines. I think that if we have a flat correction it needs to accommodate different machines and if that is not possible may it really needs to be calibrated.

Note: I tried the python script on Windows (I use Linux on a virtual machine) and found that the script doesn't work. Sorry, I should have tested this before. Normally python scripts are portable, but here the way the file is read in is different. I'll make sure that it works and tell you. I used it quite frequently, it's handy. Even better would be if someone volunteered to build a little (web) interface around it :-)

el-folie, 2022-01-25:
Yes, alpha 12.1.

And yes, the analogue hardware variation comes into play quite apparent here.

image-et-son, 2022-01-25:
[syx_converter.zip](https://github.com/image-et-son/p600fw/files/7936601/syx_converter.zip)

Hi, the script also works in Windows now. It requires Python 3.2 or higher. When you type in "python" at the command prompt it shows the version. (To leave the python script mode type "quit()".) The new script is also on GitHub.

el-folie, 2022-01-26:
Thank you - also for alpha13! Will take some time later today to dive in... looking so forward to it. :-)

image-et-son, 2022-01-29:
The topics was covered in the discussion #98 and lead to an enhancement proposel #104 .

closed 2022-01-29 by image-et-son