image-et-son/p600fw #89: Alpha 12 UI Bug: menu selection error when switching to preset mode

image-et-son, 2022-01-19:
Alpha12:
When switching from panel to preset mode the last used parameter value is still selectable by the Speed knob, even though To Tape hasn´t been pressed yet to select a parameter from the menue.

_Originally posted by @el-folie in https://github.com/image-et-son/p600fw/issues/79#issuecomment-1016868268_

image-et-son, 2022-01-20:
In my understanding this has always been like that. For example, even when switching to preset mode without "to Tape" pressed; I would expect to be able to change the seq/arp speed. I am not totally sure now, but would the alternative be the the data dials does nothing after switching to preset mode until a menu parameter is selected?

el-folie, 2022-01-20:
The more exact "problem" description is this: when in panel mode, for example, env shape is active and then switching to preset mode, the Speed knob still changes the env shape. I´d expect the Speed knob to behave like when switching on for the first time, so that it resets to arp/seq speed on the first switching to preset mode (to not alter existing presets accidentally), or, when preset mode had been already active before switching to panel mode, that the former last active parameter would still be on the Speed knob.

I don´t know why I became aware of this only now. Will check back tomorrow how v2.1rc3 behaved, with a then hopefully clear head again...

image-et-son, 2022-01-20:
I have done the following: in preset patch mode (without To Tape) the data dial defaults to clock speed but the OS remembers the last selection. I am not sure if it was that way, certainly not the default to clock speed.

I just think in several ways the expectations in terms of consistency, refinement and convenience of the user interface have slightly grown in this project :-)

el-folie, 2022-01-21:
You are right, I checked back with v2.1rc3 and it was exactly the same, parameters are being kept in memory when switching between panel and preset mode. But I think your new solution is better, to reset to just arp/seq speed on Speed as it can´t accidentally alter the preset that way.

And yes, it´s definitely an effect of your expertise! Users will become more aware of little things in the OS the smoother it gets. :-)

el-folie, 2022-01-22:
Hi, I tested the new behaviour and have to say please forget my proposal, sorry! It was more intuitive before and so it would be good to have it back like it was, so that the parameter assigned to the speed knob is always the same in preset and panel mode and doesn´t change. 

(One example: with the new change parameter mode, when switching from preset mode back to panel mode with a sequence running and having speed on the speed knob, the speed knob in panel mode then reassigns to the former panel mode parameter, which is just too counterintuitive. And this will apply to every parameter that a user potentially was fine-tuning while comparing panel with preset mode for example.)

So, back to former behaviour would be great and this "issue" is done for good as it never was an issue really. My bad!

el-folie, 2022-01-26:
Function tested & back to normal confirmed. Done.

commit cd429436 (2022-01-25): Optimization for high frequency pot pick-up (#84), fix #89
closed 2022-01-26 by image-et-son