image-et-son/p600fw #92: BUG when in MIDI management mode pressing preset causes display to go blank

matrix12x, 2022-01-21:
Alpha 12

When in MIDI management mode (from tape + record) pressing Preset causes display to go blank and it won't recover. Have  to cycle power.

Not sure if this was always the case. I hit preset, because I had forgotten how to get out of midi management mode (hit record).

image-et-son, 2022-01-23:
Hi, I just checked and in my case this action dumps the patch bank. I even checked if that works with no patches stored (a crash in in such a situation would be a classic) but that also works. It sends MIDI and scrolls a message.

matrix12x, 2022-01-24:
ugh. My crazy P600. I may have to re-flash using hex.

matrix12x, 2022-01-24:
Yeah, its weird, it was not dumping and went blank forever.

I reflashed via hex. Works fine. ok to close this nonexistent bug. Thank you for confirming.

closed 2022-01-24 by image-et-son