image-et-son/p600fw #93: Pitch Change after activating Glide

image-et-son, 2022-01-23:
When Glide = 0 and a new key is pressed and held, then turning up Glide changes the pitch of that note but no gliding effect sets in. Next notes are ok.

Also: Glide behaviour in poly mode seems a little strange, but it needs to be checked whether that has always been like that...

el-folie, 2022-01-23:
Alpha 12.1: on my P600 it´s now like this:
When Glide = 0 and a new key is pressed and held, then turning up Glide leads to shortly pitching to the last frequency of the voices being assigned to that key and back to the note actually held. It sounds like a glitch/bug. Next notes are sounding fine with Glide being applied correctly. This is the same in unison and poly mode.

My understanding is that Glide should only have an effect on the next new note, the note being held when turning up Glide should stay stable in pitch. Would you agree?

What´s strange with Glide in poly mode? To me it seems to behave correctly generally.

matrix12x, 2022-01-23:
Mine is doing the same as @el-folie, except, it only happens the first time I move from zero to a larger value.If I go to repeat this without recalling another patch, I cannot get the bug to repeat.

I think polyglide sounds fine.

image-et-son, 2022-01-23:
OK, I'll need to work on the glide. I have completed the panel layout switch (GliGli vs. SCI)- that works! But then you instantly try knob performance on glide to see what that is doing. Glide is obviously not sufficiently implemented for becoming a panel control yet.

el-folie, 2022-01-24:
Awesome that the SCI-GliGLi switch is already done and working! OMG! :)))

matrix12x, 2022-01-25:
Alpha 13
I confirm, glide seems to work well in both SCI and GLIGLI modes

closed 2022-01-26 by image-et-son