image-et-son/p600fw #94: Should the middle of the keyboard with no transpose be c3?

image-et-son, 2022-01-23:
When I implemented the new display variant for frequency A & B I needed to show the middle c key as c3 (otherwise I would have run out of positive integers at the bottom). But: the middle C is not c3 at the moment but c2. Should the whole keyboard be move up by one octave? There was an issue concerning mismatch between MIDI out and sound played #6 by el-folie.

matrix12x, 2022-01-24:
Is it possible to make the middle C3 and shift the MIDI notes accordingly? 
Also, this is not an issue for me and my use cases. 

What caused it to have to shift in the new display variant?

el-folie, 2022-01-24:
All I can say is that frequency/oscillator-footsetting-wise the P600 matches Repro5 perfectly, since you corrected the midi note number some weeks ago, so it´s already perfect like it is.

If this issue is just about what´s showing on the display for the middle c, to me that´s completely unimportant. The compatibility with Repro5 and external gear is more important. So if the values shown on LCD are not c0 to c5 but c1 to c6 or "c.1"(=c-1) to c4 would al be fine to me as long as the oscillator foot settings don´t change and so the compatibility with external gear stays intact frequency-wise.

image-et-son, 2022-01-24:
ok, I just leave it as it is

closed 2022-01-24 by image-et-son