image-et-son/p600fw #96: UI issues in alpha 13

image-et-son, 2022-01-26:
1) after load the default preset the vib amount and vib frequency are not updated unless the value is changed (is an unwanted side effect of performance  tuning in the code) --> FIXED for target 13.1

el-folie, 2022-01-27:
alpha 13.1 

Bug: 
Saved scrolling message removed, but:
- first saving after power on runs fine
- from 2nd saving on strange messages appear scrolling on the LCD when saving

Proposal:
Could this be reverted to simple only Record LED flashing/no blank LCD/no scrolling?
I´m aware that some may want more info, it´s just my craving for vintage minimalism.

image-et-son, 2022-01-27:
> alpha 13.1
>
> Bug: Saved scrolling message removed, but:
>
> * first saving after power on runs fine
> * from 2nd saving on strange messages appear scrolling on the LCD when saving
>
> Proposal: Could this be [...]

Yes, will be like that

closed 2022-01-29 by image-et-son