image-et-son/p600fw #99: Provide way to switch off MIDI to arp/seq

image-et-son, 2022-01-26:
Some people may want to play normally over the arpeggiator using external MIDI. With the recent change that external MIDI plays into the arpeggiator that is no longer possible. Therefore the user should be able to configure the behaviour.

Proposal: in "local on" mode external MIDI plays on top of the arpeggiator and in "local off" external MIDI plays into the arpeggiator. Reasoning: using the external MIDI to play into sequencer and arpeggiator is most relevant in a) the studio and/or b) when there is something wrong with the local keyboard. In the first case there is no need to support additional free play and in the second case the arpeggiator depends on eternal MIDI anyway.

el-folie, 2022-02-02:
Alpha 13.4

I can confirm the above functionality, while there´s still one problem left:

In Local Off mode the internal keyboard additionally sends notes into the arp and so into the synth engine, even without any external MIDI loop. As the notes get sent out the MIDI Output, this could cause problems with double triggering the synth engine in a MIDI loop.

image-et-son, 2022-02-03:
OK, thanks. The problem is actually more fundamental: the arp itself sends MIDI while playing. If MIDI out is reconnected to MIDI in, then the arp plays into itself. It is impossible to distinguish between MIDI notes from keyboard and notes from running arp. So there are three choices:
1) I interrupt the cycle by not sending MIDI notes from arp in local off mode (easiest but somewhat arbitrary)
2) I undo the change that the arp can be controlled by external MIDI (there might be people who need this)
3) Introduce of local on mode a variant which toggles whether external MIDI play into arp or not and never play external MIDI into arp (and sequencer) in local off mode (obviously the most versatile but more work)

el-folie, 2022-02-03:
I needed to think a moment on this. Purely from a logical viewpoint and how I know/remember local off behaviour I´d assume a behaviour like this:

Internal keyboard sends its notes as MIDI notes out of the MIDI port > MIDI loop into a computer and back to the P600 MIDI input port > MIDI gets sent to the internal arp, which plays the notes.

So, method 1. should really do as a simple solution. To avoid double MIDI notes simply different MIDI channels for sending/receiving would do.

Does this make sense or am I overlooking something?

el-folie, 2022-02-03:
alpha 13.5:
To me it´s perfect now! Arp triggers only via MIDI out>in loop, nice!

closed 2022-02-03 by image-et-son