image-et-son/p600fw #103: UI Bugs with target alpha 13.3

image-et-son, 2022-01-29:
1) When started in preset mode the loaded preset directly received the display dot as if the sound was edited

2) In preset mode changing the arp/seq speed shows up as a dot in the diplay as if the preset was changed (which it is not, since the speed is not stored in the patch)

3) In preset patch mode: when the last pot to be touched is the data pot pressing From Tape will immediately apply the current data pot value and load that patch

matrix12x, 2022-02-01:
Compiled 7079a0e
On boot into preset mode I'm still getting a "dot" in the display. (Bug)

If I enter manual mode and then go back into preset mode there is no "dot." (seems ok)

image-et-son, 2022-02-01:
That's one stubborn bug. I fixed it but then I needed to change something very close by...

el-folie, 2022-02-01:
> 

Can confirm Matrix12x´ description.

el-folie, 2022-02-01:
alpha 13.4
1, The dot when booting in preset mode seems to be gone here. Booted a few times in panel/preset mode. Confirmed.

4. The dot randomly appearing when entering preset mode is still happening. No pattern to it.

matrix12x, 2022-02-02:
I confirm that the dot is not there

closed 2022-02-02 by image-et-son