image-et-son/p600fw #107: Mod wheel doesn´t work anymore when flashing to alpha 13.3, on 13.2 it works fine

el-folie, 2022-02-01:
Funny behaviour - maybe because of performance tuning? Maybe the mod wheel scanning is so fast that the hardware on my P600 is too slow to react?

image-et-son, 2022-02-01:
Oh, sorry, I mistyped it in the code and wrote ppPitchWheel where ppModWheel  would have been correct. It is never scanned. I'll post a hotfix later today - I am on business trip at the moment.

What do you think about the general responsiveness and smoothness? Esp. pitch bender, LFO amunt, freq A & B. I was quite happy with what the latest trade off achieved. 

Only unison pushes the limits which can be heard on the frequency dials. I guess adding glide would make it a little worse again.

el-folie, 2022-02-01:
Ah, I see, haha - and I instantly interpreted the dead mod wheel as "my machine again...". Glad to hear it´s the code really. :)

I´ll test the mentioned features/responsiveness later today and answer here...

Have a good business trip! 👍

el-folie, 2022-02-01:
> What do you think about the general responsiveness and smoothness? Esp. pitch bender, LFO amunt, freq A & B. I was quite happy with what the latest trade off achieved.
> 
> Only unison pushes the limits which can be heard on the frequency dials. I guess adding glide would make it a little worse again.

Just tested it. Alpha 13.3. Poly mode:
- LFO amount is smooth and responsive, even on fast tweaks, no lag.
- freq A&B are very responsive. Any stepping on slow tweaking might just be resolution steps, no lag. (Z80 also reveals those tiny "kr-r-r-r-r-r-r-r" steps on freq parameters)
- Bender is very responsive. Any stepping might just be resolution steps, no audbile lag. Question is if the audible tiny discrete steps on slow frequency sweeps in one octave bends could be remedied by a higher PB resolution, if that wouldn´t introduce other problems of course. (But Z80 also reveals those tiny "kr-r-r-r-r-r-r-r" steps on freq parameters at high freqs)

In unison:
Yes, there is some stepping with pitch bend in one octave bends, but maybe that´s just the limit of DAC & demultiplexing speeds for all voices at once. And maybe most users wouldn´t work the pitch bend as hard and fast as I do on purpose when testing. ;-) And adding Glide doesn´t really make it worse, to the contrary, when the Glide time is long enough it smoothes out any possible stepping from pitch bend.

All in all I´d say this is an excellent state of responsiveness from the old P600 hardware. Any lag we had on early alphas is gone. Even the main volume is smooth and fast now and I remember it being laggy before. So - excellent work and you have every right to be happy about this. :)

image-et-son, 2022-02-01:
When bringing back the mod wheel I realized that this needs to be in responsive mode continuously. That puts more strain on the system. I therefore had to reduce the scanning rate for less important pots by about 20% which means these have more lag in initial response (but once touched become as responsive as any other pot). Changes will come with version 13.5

el-folie, 2022-02-01:
Thank you - mod wheel working again in alpha 13.4 confirmed.

closed 2022-02-01 by image-et-son