image-et-son/p600fw #109: Tape Sync is being influenced by the arp speed parameter, no constant triggering possible

el-folie, 2022-02-01:
When feeding the tape input with a constant ticking clock source (CR-8000, 8th or 16th) not every single clock pulse leads to an arp trigger. There are drop outs here and there, can´t see a pattern yet. 

And the arp speed parameter knob has an influence too: 
Only on speed 99 almost but not every clock pulse is converted to a trigger impulse for the arp and there are drop outs of pulses. If the speed parameter is reduced to about 80 the arp gets only triggered here and there. Further reduction of arp speed leads to only one arp trigger impulse every few beats and so on. So there must be some dependence of the arp speed and the tape input as a clock source. Maybe there once was the idea to let speed act as a clock divider, but it´s just all over the place without any reliability yet.

(Proposal: It would be nice if the arp speed parameter could act like a divider in midi sync mode, as the midi beat clock is ticking so fast that the arp becomes unusable without a divider option. In Tape Sync mode both a clock divider/multiplier would be nice to tailor the arp speed.)

(I´ll test v2.1rc3 tomorrow to see if it was always like this.)

image-et-son, 2022-02-03:
There are still corners of the GliGli firmware I have not yet worked through :-) (but not so much anymore). Clock is one such area. And I discovered, that it is already supposed to work as you describe: the speed/data pot selects a clock divider from the following list: 
{192,168,144,128,96,72,48,36,24,18,12,9,6,4,3,2}

Just a little piece of information. I now need to see where the mechanism goes wrong. Would be good if you could say something about 2.1RC3 while I concentrate on the code.

el-folie, 2022-02-03:
Yes, sorry, 2.1rc3 behaves the same! Forgot to type that after testing yesterday night.

image-et-son, 2022-02-03:
I just wanted to give you a thumbs up for that info, but it is not good news so it seemed not appropriate.

On my modular synth I have a MIDI module that can deliver all types of clock divisions. I see what I can do.

el-folie, 2022-02-03:
Great! I´ll provide a little recording of the weird "sync" behaviour of P600 to CR-8000 16th trigger pulses shortly...

el-folie, 2022-02-03:
[P600_typeIN_syncByPulse.zip](https://github.com/image-et-son/p600fw/files/7998354/P600_typeIN_syncByPulse.zip)

image-et-son, 2022-02-03:
I provided alpha 13.6. I have not posted it on Gearspace - the fast cycles may confused people and put them off as it looks as if something needs fixing all the time (whereas in reality we're cleaning up). Clock should be working!

matrix12x, 2022-02-03:
Alpha 13.6, 
just tried it out and it sounds good. Now my workable range seems to be about double. I can get to about 40 or so with it still sounding nice and vibrato-ish.

I like how you set VCO to half and kept VCA at full.

image-et-son, 2022-02-03:
> I like how you set VCO to half and kept VCA at full.

Unfortunately only for vibrato. For LFO is gets mixed up with the other targets and there I cannot separate it to affect only the VCA stronger - at least not without significant refactoring.

el-folie, 2022-02-04:
imogen, you are a magician - Tape Sync works perfect now! Haha, unbelievable...

Re. the new halved VCO depth mod wheel, I like 13.5 better and I´d like even more LFO depth (max. VCO mod depth, from infrasound to ultrasound) on the mod wheel for some ring mod type sounds that use poly mod to filter for example. With Inital Amount I can set a tiny depth to modulate the sound only a bit and by playing live with the mod wheel I can get the sound to take off into the stratosphere... ;-) I think to still ensure a broad range of use cases a lo/mid/hi switch for mod wheel depth would be best. (While Initial Amount could always deliver whole range depth).

matrix12x, 2022-02-04:
@el-folie I agree, I think we should bring back min/lo/mid/high.

el-folie, 2022-02-04:
And we also have the 2nd LFO = Vibrato, which always had a somewhat lower depth for normal vibrato purposes IIRC? Maybe the Vibrato LFO could be made more attractive by targets like VCO AB/A/B, VCA, VCF too. Then, the 1st LFO would be free for more extreme stuff...

matrix12x, 2022-02-04:
Good point. but I think this mod should happen after the current alpha is shored up

el-folie, 2022-02-04:
Definitely...

matrix12x, 2022-02-04:
The code for this mod (additional targets for vibe) can easily be copied form my old code same with vibe waveforms.

closed 2022-02-04 by image-et-son